RavenSystem/esp-homekit-devices #921: problem with haa firmware to flash a sonoff rf r2

lorenzo677, 2020-05-02:
Hi everyone 
I have a problem with flashing a Sonoff rf r2 with HAA firmware. I've tried the procedure for flashing from the terminal and from NodeMCU PyFlasher (both processes seem to have a positive ending)  and the result was the same: the Sonoff seems tho be dead, neither wifi network nor the light is turned on. If I repeat the same procedure with Tasmota, it works instead.
I don't really know how to put HAA firmware in this sonoff. How should I proceed?

pilot1981, 2020-05-02:
This is very strange: it must works! You soldered pin on GPIO? Check soldering points!

lorenzo677, 2020-05-02:
yes i soldered pin on GPIO and they seems to be well soldered

pilot1981, 2020-05-02:
I used HAA on R3 RF and all gone well!
I don’t understand...you used last haaboot?

lorenzo677, 2020-05-02:
Sincerely I've tried with every version i've found, but i've obtained always the same result. i don't really know where i do wrong

lorenzo677, 2020-05-02:
I test also the Sonoff with putty and this is the result [putty.txt](https://github.com/RavenSystem/esp-homekit-devices/files/4568003/putty.txt)

closed 2020-06-01 by stale[bot]

pilot1981, 2020-06-01:
Sonoff basic R3 RF work fine with HAA