RavenSystem/esp-homekit-devices #922: A way to enter setup mode

nikbyte, 2020-05-02:
Hi!

Is there any way to enter into setup mode if I set wrong I/O inputs in my json configuration?
I flashed device without disassemble (OTA) and this is too hard to disassemble it and solder wires to reflash it, I have a brick which connects to my wifi but this is not in setup mode.

RavenSystem, 2020-05-02:
Emergency setup mode:
Plug it. 
Unplug after 1 second. 
Plug it. 
Device is in setup mode. Connect to it usong it IP address.

nikbyte, 2020-05-02:
I tried many times. It disconnects from wifi and then connects to wifi. But still connection refused. What am I doing wrong? Firmware 2.1.1.

RavenSystem, 2020-05-02:
Maybe you unplug it too soon. Try with 2 seconds instead 1.

RavenSystem, 2020-05-02:
Or add it to HomeKit and turn on and off accessory 8 times quickly.

nikbyte, 2020-05-02:
I don't know why, but I can't add it to homekit, the device isn't accessible.
But after switching on it gets IP address via DHCP and pingable.

RavenSystem, 2020-05-02:
The installation is unfinished.

RavenSystem, 2020-05-02:
Keep it connected for at least 15 minutes.

nikbyte, 2020-05-02:
Hmmm. I was waiting for about an hour after initial flash.
(I was doing this with another devices and everything was ok, but this one (new type of device) – doesn't work).

nikbyte, 2020-05-02:
Ok. Anyway I've got what do to. Thank you very much for your advices!

RavenSystem, 2020-05-02:
You can view HAA OTA installation logs using this: https://github.com/RavenSystem/esp-homekit-devices/wiki/Build-Instructions#haa-ota-logs

nikbyte, 2020-05-02:
Wow. Interesting.

NEW CONNECTION LocalPort=dd9a DNS IP: 13.236.229.21 local..OK remote..OK SSL..OK set_fd to github.com port 443..OK
sent OK
HTTP returns 404
Get file
DOWNLOADING
GET /RavenSystem/haa_ota/releases/download/3.0.2/main.bin HTTP/1.1
Host: github.com

nikbyte, 2020-05-02:
It seems, there are no main.bin and main.bin.sig in 3.0.2

nikbyte, 2020-05-02:
More logs

STARTING

Get version RavenSystem/haa_ota
NEW CONNECTION LocalPort=cfd5 DNS IP: 140.82.118.3 local..OK remote..OK SSL..OK set_fd to github.com port 443..Get version RavenSystem/haa_ota
NEW CONNECTION LocalPort=d08e DNS IP: 140.82.118.3 local..OK remote..OK SSL..OK set_fd to github.com port 443..OK

GET /RavenSystem/haa_ota/releases/latest HTTP/1.1
Host: github.com

HTTP returns 302 for Current ROM is 0
Compare
3.0.2 with 2.2.3 = 1
RavenSystem/haa_ota@version:"3.0.2"
Current ROM is 0
Running BOOT
Get hash
DOWNLOADING
GET /RavenSystem/haa_ota/releases/download/3.0.2/main.bin.sig HTTP/1.1
Host: github.com

NEW CONNECTION LocalPort=cfd6 DNS IP: 140.82.118.3 local..OK remote..OK

GET /RavenSystem/haa_ota/releases/latest HTTP/1.1
Host: github.com

OK SSL..OK set_fd to github.com port 443..HTTP returns 302 for Current ROM is 0
Compare
3.0.2 with 2.2.3 = 1
RavenSystem/haa_ota@version:"3.0.2"
Current ROM is 0
Running BOOT
Get hash
DOWNLOADING
GET /RavenSystem/haa_ota/releases/download/3.0.2/main.bin.sig HTTP/1.1
Host: github.com

NEW CONNECTION LocalPort=d08f DNS IP: 140.82.118.3 local..OK remote..OK SSL..OK set_fd to github.com port 443..OK
sent OK
HTTP returns 404
Get file
DOWNLOADING
GET /RavenSystem/haa_ota/releases/download/3.0.2/main.bin HTTP/1.1
Host: github.com

NEW CONNECTION LocalPort=cfd7 DNS IP: 140.82.118.3 local..OK remote..OK
sent OK
OK SSL..OK set_fd to github.com port 443..HTTP returns 404
Get file
DOWNLOADING
GET /RavenSystem/haa_ota/releases/download/3.0.2/main.bin HTTP/1.1
Host: github.com

NEW CONNECTION LocalPort=d090 DNS IP: 140.82.118.3 local..OK remote..OK SSL..OK set_fd to github.com port 443..OK
sent OK
HTTP returns 404

nikbyte, 2020-05-02:
I have also another devices which I flashed maybe two weeks ago, but never switched on, they can't install too with the same log.

nikbyte, 2020-05-03:
It seems I was ok while HAA OTA v2.2.3 was latest. After HAA OTA v3.0.0 there are no main.bin and I can't finish install of my devices. Could you please be so kind and add main.bin/main.bin.sig there?

RavenSystem, 2020-05-03:
You are using an old haaboot.bin. Please, use last version for new installations.

nikbyte, 2020-05-03:
Yes. I'll use for new installations new loader, but could you please place these files for me to upgrade existing devices which I can't reflash without disassemple?

RavenSystem, 2020-05-03:
Sorry, it is incompatible with current versions and it breaks new installations.

nikbyte, 2020-05-03:
Ok. I'm trying to workaround this with fake server.
I placed files and made required redirects. Maybe you have idea why this error occurs?

```
STARTING

Get version RavenSystem/haa_ota
NEW CONNECTION LocalPort=ce04 DNS IP: 10.11.12.2 local..OK remote..OK SSL..OK set_fd to github.com port 443..OK

GET /RavenSystem/haa_ota/releases/latest HTTP/1.1
Host: github.com

HTTP returns 302 for Current ROM is 0
Compare
2.2.3 with 2.2.3 = 0
RavenSystem/haa_ota@version:"2.2.3"
Current ROM is 0
Running BOOT
Get hash
DOWNLOADING
GET /RavenSystem/haa_ota/releases/download/2.2.3/main.bin.sig HTTP/1.1
Host: github.com

NEW CONNECTION LocalPort=ce05 DNS IP: 10.11.12.2 local..OK remote..OK SSL..OK set_fd to github.com port 443..OK
sent OK
HTTP returns 200
Get file
DOWNLOADING
GET /RavenSystem/haa_ota/releases/download/2.2.3/main.bin HTTP/1.1
Host: github.com

NEW CONNECTION LocalPort=ce06 DNS IP: 10.11.12.2 local..OK remote..OK SSL..OK set_fd to github.com port 443..OK
sent OK
failed, return [-0x1]
wolfSSL_send error = -125
```

RavenSystem, 2020-05-03:
This error occurs because you are trying to do a MITM attack against an encrypted connection.

nikbyte, 2020-05-03:
It checks certificate?

Help me to workaround this please, I have more than 20 bricked devices. :(

nikbyte, 2020-05-03:
I have an idea. I just must make cert where cn=github.com

I hope it doesn’t check root certificates?

nikbyte, 2020-05-03:
Doesn’t work.
Let me explain what am I doing.

I replaced github.com DNS with IP of my local server and placed files there.

I see it loads well signature, but it can’t load bin file. I think the issue isn’t in my ssl.

125 error in wolfssl doc means “out of buffer”. why? files are original, from your repository.

nikbyte, 2020-05-03:
https://www.wolfssl.com/forums/topic1488-erro-wolfsslconnect-renturn-125.html

nikbyte, 2020-05-03:
I reduced server side ssl buffers and now I don't have 125 error.
However I see in loop how it receives files. Why?

```
STARTING

Get version RavenSystem/haa_ota
NEW CONNECTION LocalPort=c294 DNS IP: 10.0.0.2 local..OK remote..OK SSL..OK set_fd to github.com port 443..OK

GET /RavenSystem/haa_ota/releases/latest HTTP/1.1
Host: github.com

HTTP returns 302 for Current ROM is 0
Compare
2.2.3 with 2.2.3 = 0
RavenSystem/haa_ota@version:"2.2.3"
Current ROM is 0
Running BOOT
Get hash
DOWNLOADING
GET /RavenSystem/haa_ota/releases/download/2.2.3/main.bin.sig HTTP/1.1
Host: github.com

NEW CONNECTION LocalPort=c295 DNS IP: 10.0.0.2 local..OK remote..OK SSL..OK set_fd to github.com port 443..OK
sent OK
HTTP returns 200
Get file
DOWNLOADING
GET /RavenSystem/haa_ota/releases/download/2.2.3/main.bin HTTP/1.1
Host: github.com

NEW CONNECTION LocalPort=c296 DNS IP: 10.0.0.2 local..OK remote..OK SSL..OK set_fd to github.com port 443..OK
sent OK
HTTP returns 200
```

nikbyte, 2020-05-03:
```
10.0.0.237 - - [03/May/2020:15:46:08 +0300] "GET /RavenSystem/haa_ota/releases/download/2.2.3/main.bin HTTP/1.1" 200 45822 "-" "-" "-"
10.0.0.37 - - [03/May/2020:15:46:09 +0300] "GET /RavenSystem/haa_ota/releases/download/2.2.3/main.bin HTTP/1.1" 200 411648 "-" "curl/7.64.1" "-"
```

when I download file (curl), I see full length of file – 411648.
Device receives only 45822 bytes. Is this normal?

nikbyte, 2020-05-03:
Ok, thank you for your all advices.

RavenSystem, 2020-05-03:
I added a transition firmware to allow old haaboot to work. Try again.

nikbyte, 2020-05-03:
Thank you very much. Your solution works perfectly. My device successfully upgraded into haa 2.1.2 and haa_ota 3.0.2. How long this will be available? I mean, when you'll make a new release, will it be supported or until next release only?

shanghaix, 2020-05-04:
I had the exact same problem, flashed some Gosund WP5 outlets with HAA using Tuya-Convert, devices went into setup mode the first time and then connected to wifi but no setup mode the second time and not visible in HomeKit and they were unresponsive and seemed bricked, until last night when I tried again and it magically started updating them when looking at the logs with the nc command. Thanks a lot @RavenSystem for fixing it, I was about to trash them and buy new ones...

nikbyte, 2020-05-04:
@RavenSystem could you please answer my question? How long this transition firmware will be available?

i3laze, 2020-05-11:
We faced same main.bin thing in another issue.
Here is dev’s answer so far:
https://github.com/RavenSystem/esp-homekit-devices/issues/683#issuecomment-625951730

blimpmason, 2020-05-15:
I maybe having a similar issue — I flashed this [Monoprice/Tuya Stitch smart outlet](https://www.monoprice.com/product?p_id=27937) using tuya-convert with this [thirdparty.bin](https://github.com/ct-Open-Source/tuya-convert/files/3849141/thirdparty.bin.zip). 

The flash was successful and I was able to connect to the device at its SSID (LCM-XXXXXX) and set the wifi details per the instructions. 

But I am never able to add the device in the iOS Home app — I've tried many times and it always times out.

I tried monitoring the logs via `nc -kulnw0 45678` and after unplugging/replugging, I just get a variations on this over and over again:

```user-init-start
user-init-done
wifiready-done
--- ota_boot...0
OTABOOT VERSION: 1.0.0
--- ota_init
userbeta='0' otabeta='0'
active_sector: 0x0
--- ota_set_verify...OFF
--- DNS: done!
--- ota_get_version
--- ota_connect LocalPort=fd5e DNS IP:192.30.255.112 local..OK remote..OK SSL..OK set_fd to github.com port 443..OK
GET /HomeACcessoryKid/life-cycle-manager/releases/latest HTTP/1.1
Host: github.com

Is there any hope for the device? I am unable to get it into setup mode or browse to its IP address, though I do see it has an IP address via my router admin (matching the MAC address).

tonysprenk, 2020-05-16:
@blimpmason where did you find the link to thirdparty.bin? It has old firmware that is broken and should be removed. I’m sorry to say that the only way to get your device working now is by flashing it by wire.

blimpmason, 2020-05-16:
@tonysprenk thanks a lot for your response! That's what I was afraid of. The thirdparty.bin I used is actually the one you posted over in this tuya-convert issue thread: https://github.com/ct-Open-Source/tuya-convert/issues/389#issuecomment-554106217

It's also linked to in this guide: https://homekit.blog/forum/flashing-tuya-devices-supported-native-with-homekit-no-need-for-bridge/

I don't have the ambition to open it up and attempt a manual flash — could I donate it to you or @RavenSystem for testing/reflashing (if either of you are in a country with US 120V outlets)? Happy to ship the smart plug if it would be useful to help others flash these devices in the future.

tonysprenk, 2020-05-16:
Hi @blimpmason,

I’m sorry that a file I made has bricked your device. I thought I had deleted all reference to the file. I have deleted the comment and asked the website to delete the blog post. Thanks for bringing it to my attention. 

I have no use for a 120v device as I live in Europe. I think it’s the same for @RavenSystem. In any case we know what’s wrong with the firmware so don’t need the device for that purpose anyway. 

If you want to flash another device with esp-HomeKit-devices the latest release can be found [here](https://github.com/RavenSystem/esp-homekit-devices/wiki/Installation).

Again sorry for your loss (😉).

RavenSystem, 2020-05-16:
If GitHub reverts headers changes, it will work.

nikbyte, 2020-05-16:
@RavenSystem when I tried to recover my device, I made fake “github” local server which just redirects to real github. As HAA_OTA don’t check root CA, it worked. I think this resolvable the same way. If people needs it, I can flash old firmware, test and write manual how to do this.

tonysprenk, 2020-05-16:
@nikbyte Yes! Please share how you did it! It would save a lot of hassle!

nikbyte, 2020-05-16:
@tonysprenk what exactly version of firmware must I flash to test this?

blimpmason, 2020-05-16:
@tonysprenk no worries at all! I knew the risks going in and obviously didn't do all the homework I should have. I take full responsibility. 

@RavenSystem @nikbyte so there might be hope for me to recover my device OTA?

tonysprenk, 2020-05-16:
Anything before February 29. I believe [this](https://github.com/RavenSystem/esp-homekit-devices/releases/tag/HAAOTA_2.1.0) is the latest broken HAA OTA.

nikbyte, 2020-05-16:
@tonysprenk if I get files from this comment, this is what we need?
https://github.com/ct-Open-Source/tuya-convert/issues/389#issuecomment-553721234

blimpmason, 2020-05-16:
@nikbyte here's the exact thirdparty.bin I used:
[thirdparty.bin.zip](https://github.com/RavenSystem/esp-homekit-devices/files/4639244/thirdparty.bin.zip)

tonysprenk, 2020-05-16:
I deleted that comment to prevent people from bricking their devices. Found a backup. Can you send me a private message on discord so I can get it to you?

tonysprenk, 2020-05-16:
Discord @tsprenk

nikbyte, 2020-05-16:
@tonysprenk But I still see comment from the link above and can download the files.
I'll try with .bin from @blimpmason

tonysprenk, 2020-05-16:
Ah i see. Yes it’s the same.

nikbyte, 2020-05-16:
@tonysprenk @blimpmason help me, I never used so old firmware.
What I must set in OTA repository and OTA binary fields? Must I enable using of pre-releases?
By default there is "HomeACcessoryKid/ota-demo" and "main.bin".

blimpmason, 2020-05-16:
@nikbyte I'm afraid I can't help — I didn't package the .bin myself, I just flashed it OTA using tuya-convert.

tonysprenk, 2020-05-16:
Ota repository is “RavenSystem/haa” Ota binary stays main.bin. But it won’t install because it cannot connect to the server due to header capitalization issue.

tonysprenk, 2020-05-16:
I guess don’t enable prerelease.

nikbyte, 2020-05-16:
yes, @tonysprenk, I understand the issue. I think my method will help, but let me test and report here.

nikbyte, 2020-05-16:
looking on that old OTA, I appreciate more and more @RavenSystem 's work.

tonysprenk, 2020-05-16:
Thanks for looking into it @nikbyte. And yes you are right without @RavenSystem our lives would be a lot less simple...

tonysprenk, 2020-05-16:
I’m afk now. If you have anymore questions I’m glad to help again in the morning. Thanks again!

nikbyte, 2020-05-17:
@blimpmason, I tested your .bin. You flashed https://github.com/HomeACcessoryKid/life-cycle-manager of version 1.0.0 . It checks ssl certificate of github and my method doesn't work to spoil it. I think only way is to solder and reflash your device by wire. Or... wait for a miracle when github.com change "location" header into "Location" because old firmwares were sensitive to register of this header and when github did this, all old firmwares lost possibility to download updates.

blimpmason, 2020-05-17:
@nikbyte, thanks so much for taking the time to test that .bin! I'll hold onto the device in case I feel inspired to solder / flash by wire. If I did want to go that route, which instructions would you recommend?

tonysprenk, 2020-05-17:
@nikbyte when did you encounter your update issue? Was it with firmware from before 29th of February? Could you share your fake server method so I can try on some of my devices that won’t update?

peros550, 2020-05-17:
Guys, please share your findings on the solution of using a local github server. 

I also have a few smart bulbs which cannot open to sold wires. 

Any idea if that works idea will work on devices running older version of LCM instead of HAA OTA?

[15 more replies]

closed 2020-05-19 by RavenSystem