RavenSystem/esp-homekit-devices #925: Help with config - Sonoff T2 EU 3

quartist, 2020-05-04:
Hello,

Could you please help me with create configuration for Sonoff T2 EU 3 like
button + button + stateless button (with 3functions)
https://github.com/RavenSystem/esp-homekit-devices/wiki/Stateless-Button

i3laze, 2020-05-14:
Hi and welcome.

Try the online HAA configurator:
https://glumb.github.io/haa-configurator/

1. Use a template for Sonoff T2 EU 3 Gangs.
2. Leave digital inputs For each accessory as is - they are the buttons.
3. Optionally, modify Digital outputs to other desired Actions:
https://github.com/RavenSystem/esp-homekit-devices/wiki/Switches
4. Copy the resulting JSON.

quartist, 2020-06-04:
Hello,

this code working:
{
  "c": {
    "f": 20,
    "l": 13,
    "n": "Test",
    "b": [
      {
        "g": 0,
        "t": 5
      }
    ]
  },
  "a": [
    {
      "0": {
        "r": [
          {
            "g": 12
          }
        ]
      },
      "1": {
        "r": [
          {
            "g": 12,
            "v": 1
          }
        ]
      },
      "b": [
        {
          "g": 0
        }
      ],
      "s": 0
    },
    {
      "0": {
        "r": [
          {
            "g": 5,
            "v": 0
          }
        ]
      },
      "1": {
        "r": [
          {
            "g": 5,
            "v": 0
          }
        ]
      },
      "2": {
        "r": [
          {
            "g": 5,
            "v": 0
          }
        ]
      },
      "t": 3,
      "b": [],
      "s": 0,
      "f0": [
        {
          "g": 9,
          "t": 1
        }
      ],
      "f1": [
        {
          "g": 9,
          "t": 2
        }
      ],
      "f2": [
        {
          "g": 9,
          "t": 3
        }
      ]
    },
    {
      "0": {
        "r": [
          {
            "g": 4,
            "v": 0
          }
        ]
      },
      "1": {
        "r": [
          {
            "g": 4,
            "v": 0
          }
        ]
      },
      "2": {
        "r": [
          {
            "g": 4,
            "v": 0
          }
        ]
      },
      "t": 3,
      "b": [],
      "s": 0,
      "f0": [
        {
          "g": 10,
          "t": 1
        }
      ],
      "f1": [
        {
          "g": 10,
          "t": 2
        }
      ],
      "f2": [
        {
          "g": 10,
          "t": 3
        }
      ]
    }
  ]
}

but two problems:
- when I using stateless button, led don't working? Not blinking when touch.
- Its really difficult to hit "double button". Needs more precision between hits.

i3laze, 2020-06-04:
Please post flat JSON next time)

According to [this pinout](https://templates.blakadder.com/sonoff_T1_eu_2.html)LED is GPIO13.
Add it as an output for 0,5sec for both of relays. Probably you need also invert checkbox.

closed 2020-07-05 by stale[bot]