RavenSystem/esp-homekit-devices #928: Devices don't reconnect after no electricity at home

pkordal, 2020-05-06:
Hi @RavenSystem. At first I would like to say how much I value your work with HAA. I started with Homekit few months ago and now I have more than 15 devices (mostly Shelly and Sonoff) using your firmware. 

Although I have some issue with devices which happens after electricity outage I suffer from time to time at my home. 

My home WiFi use to start slowly and sometimes I need to reset it. 
In such situation few of HAA devices (they are on newest version) do not reconnect to WiFi. What helps is enabling setup mode with 8 time toggle. Device restarts, I can connect to it with browser, press Save and after another restart it gets fully functional. 

Would not it be possible to make device check its state and restart in case there is problem with connection to WiFi?

Thanks
Paweł

morphix9, 2020-05-06:
Pawel, I'm glad that you opened up this, I'm having the exact same problem and it's driving me crazy for almost a month. I was struggling to open up the issue because I thought it's something on my end (wi-fi, devices, etc.). I also have no logs because I can't access those by connecting to the shelly via USB (I don't know what I'm doing wrong, I've checked the wiki page), only OTA logs.

I have more than 20 devices (Shelly's 1 and Sonoff 4ch) and those on more recent HAA (1.9.1 or never) are not coming back after a short electricity outage. I have updated to 2.1.2 a few days ago, but it hasn't fixed the issue.

morphix9, 2020-05-07:
This is the exact same issue I’m having: https://github.com/maximkulkin/esp-homekit/issues/125

i3laze, 2020-05-09:
Can’t confirm, guys. Mine Shellys and Sonoff Basics do come back eventually. 

Do you have multiple WiFi Access Points or just one?
Serial log output is welcome.

morphix9, 2020-05-11:
- I have multiple wi-fi access points, yes. I've tried with "Normal" and "Force SSID" setting, but can't tell if this is the reason. Fact is that after the electricity is back, wi-fi needs few minutes to be established, EPS needs just a few seconds.

- I can't get the full log. When I try to connect to such "stuck" device via USB (`screen` command) I'm getting flooded with messages as described in this issue maximkulkin/esp-homekit#125:

```
[W]sec XXX error
[W]sec XXX error
[W]sec XXX error
[W]sec XXX error
...
```

pkordal, 2020-05-11:
> Fact is that after the electricity is back, wi-fi needs few minutes to be established, EPS needs just a few seconds.

I can say my situation is very much the same. 

It might (I am guessing a bit) be that some of my 3 access points gets up earlier than router. It would make sonoffs connect to WiFi but not proper network. And once router is up they cannot "find themselves"

What is worth mentioning: When devices are all connected to the WiFi and I switch WiFi off and after few minutes on they are all back.

morphix9, 2020-05-12:
@pkordal Are you also experience that erasing flash on those devices happens in less than a second (see related issue above)? It looks like I can't erase the flash completely, because sometimes after erasing flash, uploading new OTA/HAA, it happens that the device is already set up in Homekit (like it would not delete the Homekit pairing ID).

mat130780, 2020-05-13:
I have just flashed HAA on a Sonoff basic a few days ago to use it as a garage door opener.
It works fine once it is connected, but I also have issues with reconnecting after a power outage or after I restart my wifi router. HAA will not reconnect then. 
Besides that, HAA is a great piece of software, but this problem really annoys me.

morphix9, 2020-05-18:
> I have just flashed HAA on a Sonoff basic a few days ago to use it as a garage door opener.
> It works fine once it is connected, but I also have issues with reconnecting after a power outage or after [...]

This issue is making me think about migrating away from HAA, as it's a pain to re-flash devices after (almost) every power outage ;(

buarle, 2020-05-21:
same here. 2 out of 3 shelly 1 did not reconnect after a power outage yesterday. all were running version 1.7. Since those shellys are mounted behind lightswitches, it's a lot of work to get them towards setup mode again...

Any suggestions how to adress this?

Thanks and best regards,
buarle

serkanyilmaz, 2020-05-21:
> This issue is making me think about migrating away from HAA, as it's a pain to re-flash devices after (almost) every power outage ;(

Hi is there an alternative to HAA?

morphix9, 2020-05-21:
> same here. 2 out of 3 shelly 1 did not reconnect after a power outage yesterday. all were running version 1.7. Since those shellys are mounted behind lightswitches, it's a lot of work to get them towards setup mode again...
> 
> Any suggestions how to adress this?
> 
> Thanks and best regards,
> buarle

How do you power those Shelly's? 220V or 24V?
Do you have multiple wi-fi access points? Are you using Normal or Force BSSID mode?

buarle, 2020-05-21:
> How do you power those Shelly's? 220V or 24V?
> Do you have multiple wi-fi access points? Are you using Normal or Force BSSID mode?

230V (Germany) and only one Wi-Fi from the router (Fritzbox). The devices act in normal mode.

morphix9, 2020-05-21:
I can't find a pattern, I have so many devices, but just some are dying. It's happening also with the very recent version of HAA (today again a short outage). Some old devices with HAA 
1.5.4 and 1.8.2 are working like a charm.

RavenSystem, 2020-06-06:
Maybe last version 2.4.5 could fix `[W]sec XXX error`.

morphix9, 2020-06-08:
Ok, will update and let you know if it's happening again. Thanks Jose!

closed 2020-07-08 by stale[bot]