RavenSystem/esp-homekit-devices #930: cannot setup wifi by web interface

5high, 2020-05-09:
I am using ESP-01 flash the firmware and try to setup wifi on web UI but nothing happened.

i have captured com port log below:

'''
Using 'COM9' as serial port.
Showing logs:
[20:18:24]rl[20:18:24]pm_task_hdl : 3ffef410, prio:1, stack:176
[20:18:24]frc2_timer_task_hdl:0x3fff4b48, prio:12, stack:200
[20:18:24]
[20:18:24]ESP-Open-SDK ver: 0.0.1 compiled @ Apr 25 2020 18:20:48
[20:18:24]phy ver: 273, pp ver: 8.3
[20:18:24]
[20:18:24]
[20:18:24]
[20:18:24]
[20:18:24]! Sysparam -4
[20:18:24]WiFi Init
[20:18:24]No WiFi config found
[20:18:24]
[20:18:24]HAA OTA - SETUP MODE
[20:18:24]
[20:18:24]Start AP SSID=HAA-6807BC
[20:18:24]Start DHCP server
[20:18:24]mode : sta(b4:e6:2d:68:07:bc) + softAP(b6:e6:2d:68:07:bc)
[20:18:24]add if0
[20:18:24]add if1
[20:18:24]bcn 100
[20:18:24]Start WiFi scan
[20:18:24]Start DNS server
[20:18:24]Start HTTP server
[20:18:26]scandone
[20:18:28]add 1
[20:18:28]aid 1
[20:18:28]station: e4:a7:a0:ac:7e:7a join, AID = 1
[20:18:29]lwip_read 64
[20:18:29]! Unknown: GET /connecttest.txt
[20:18:29]lwip_read 47
[20:18:29]Unknown
[20:18:29]Redirecting to http://192.168.4.1/settings
[20:18:29]lwip_read 0
[20:18:29]Client disconnected
[20:18:29]lwip_read 64
[20:18:29]! Unknown: GET /connecttest.txt
[20:18:29]lwip_read 47
[20:18:29]Unknown
[20:18:29]Redirecting to http://192.168.4.1/settings
[20:18:29]lwip_read 0
[20:18:29]Client disconnected
[20:18:33]lwip_read 64
[20:18:33]! Unknown: GET /res/static/thirdparty/connect.jpg?t=1589026713
[20:18:33]lwip_read 64
[20:18:33]lwip_read 64
[20:18:33]lwip_read 64
[20:18:33]lwip_read 64
[20:18:33]lwip_read 64
[20:18:33]lwip_read 64
[20:18:33]lwip_read 33
[20:18:33]Unknown
[20:18:33]Redirecting to http://192.168.4.1/settings
[20:18:33]lwip_read 0
[20:18:33]Client disconnected
[20:18:34]lwip_read 64
[20:18:34]lwip_read 64
[20:18:34]lwip_read 7
[20:18:34]lwip_read 0
[20:18:34]Client disconnected
[20:18:36]scandone
[20:18:46]scandone
[20:18:48]lwip_read 64
[20:18:48]! Unknown: POST /api/filediff?block_list=1&clienttype=8&channel=00000000000
[20:18:48]lwip_read 64
[20:18:48]! Unknown: POST 000000000000000000000&version=6.9.5.1&devuid=BDIMXV2%2DO%5FA7367
[20:18:48]lwip_read 64
[20:18:48]! Unknown: POST 0A9CC6845BC8BEDB132E53360A8%2DC%5F0%2DD%5FACE4%5F2E00%5F9541%5FF
[20:18:48]lwip_read 64
[20:18:48]! Unknown: POST 544%2E%2DM%5FC85B76835DE9%2DV%5F02205ECB&rand=8447cddc85376e6748
[20:18:48]lwip_read 64
[20:18:48]! Unknown: POST a0498441d270fba64b4e4b&time=1589026728&vip=0&wp_retry_num=2&logi
[20:18:48]lwip_read 64
[20:18:48]! Unknown: POST d=JwAxADUAOAA5ADAAMgA2ADcAMgA4ACwAMQAwAC4AMQAwAC4AMQAwAC4AMQA4AD
[20:18:48]lwip_read 64
[20:18:48]! Unknown: POST AALAA0ADIA%0ANAA4ACwANAAzADEANgAnAA%3D%3D
[20:18:48]lwip_read 64
[20:18:48]lwip_read 64
[20:18:48]lwip_read 64
[20:18:48]lwip_read 64
[20:18:48]lwip_read 64
[20:18:48]lwip_read 64
[20:18:48]lwip_read 64
[20:18:48]lwip_read 64
[20:18:48]lwip_read 64
[20:18:48]lwip_read 64
[20:18:48]lwip_read 64
[20:18:48]lwip_read 64
[20:18:48]lwip_read 64
[20:18:48]lwip_read 64
[20:18:48]lwip_read 64
[20:18:48]lwip_read 64
[20:18:48]lwip_read 33
[20:18:48]Unknown
[20:18:48]Redirecting to http://192.168.4.1/settings
[20:18:48]lwip_read 0
[20:18:48]Client disconnected
[20:18:49]lwip_read 64
[20:18:49]lwip_read 64
[20:18:49]lwip_read 37
[20:18:49]lwip_read 0
[20:18:49]Client disconnected
[20:18:53]lwip_read 64
[20:18:53]! Unknown: GET /connecttest.txt
[20:18:53]lwip_read 47
[20:18:53]Unknown
[20:18:53]Redirecting to http://192.168.4.1/settings
[20:18:55]lwip_read -1
[20:18:55]Client disconnected
[20:18:56]scandone
[20:18:57]lwip_read 64
[20:18:57]! Unknown: GET /connecttest.txt
[20:18:57]lwip_read 47
[20:18:57]Unknown
[20:18:57]Redirecting to http://192.168.4.1/settings
[20:18:57]lwip_read 0
[20:18:57]Client disconnected
[20:19:04]lwip_read 64
[20:19:04]! Unknown: GET /res/static/thirdparty/connect.jpg?t=1589026744
[20:19:04]lwip_read 64
[20:19:04]lwip_read 64
[20:19:04]lwip_read 64
[20:19:04]lwip_read 64
[20:19:04]lwip_read 64
[20:19:04]lwip_read 64
[20:19:04]lwip_read 33
[20:19:04]Unknown
[20:19:04]Redirecting to http://192.168.4.1/settings
[20:19:04]lwip_read 0
[20:19:04]Client disconnected
[20:19:05]lwip_read 64
[20:19:05]lwip_read 64
[20:19:05]lwip_read 7
[20:19:06]scandone
[20:19:07]lwip_read 0
[20:19:07]Client disconnected
[20:19:07]lwip_read 64
[20:19:07]! Unknown: POST /api/filediff?block_list=1&clienttype=8&channel=00000000000
[20:19:07]lwip_read 64
[20:19:07]! Unknown: POST 000000000000000000000&version=6.9.5.1&devuid=BDIMXV2%2DO%5FA7367
[20:19:07]lwip_read 64
[20:19:07]! Unknown: POST 0A9CC6845BC8BEDB132E53360A8%2DC%5F0%2DD%5FACE4%5F2E00%5F9541%5FF
[20:19:07]lwip_read 64
[20:19:07]! Unknown: POST 544%2E%2DM%5FC85B76835DE9%2DV%5F02205ECB&rand=5428e371175c0883c9
[20:19:07]lwip_read 64
[20:19:07]! Unknown: POST ec74d3d1ea2b0a89845aa6&time=1589026747&vip=0&wp_retry_num=2&logi
[20:19:07]lwip_read 64
[20:19:08]! Unknown: POST d=JwAxADUAOAA5ADAAMgA2ADcANAA3ACwAMQAwAC4AMQAwAC4AMQAwAC4AMQA4AD
[20:19:08]lwip_read 64
[20:19:08]! Unknown: POST AALAA0ADIA%0ANAA4ACwANwAwADAAOAAnAA%3D%3D
[20:19:08]lwip_read 64
[20:19:08]lwip_read 64
[20:19:08]lwip_read 64
[20:19:08]lwip_read 64
[20:19:08]lwip_read 64
[20:19:08]lwip_read 64
[20:19:08]lwip_read 64
[20:19:08]lwip_read 64
[20:19:08]lwip_read 64
[20:19:08]lwip_read 64
[20:19:08]lwip_read 64
[20:19:08]lwip_read 64
[20:19:08]lwip_read 64
[20:19:08]lwip_read 64
[20:19:08]lwip_read 64
[20:19:08]lwip_read 64
[20:19:08]lwip_read 34
[20:19:08]Unknown
[20:19:08]Redirecting to http://192.168.4.1/settings
[20:19:08]lwip_read 0
[20:19:08]Client disconnected
[20:19:08]lwip_read 64
[20:19:08]lwip_read 64
[20:19:08]lwip_read 37
[20:19:08]lwip_read 0
[20:19:08]Client disconnected
[20:19:08]lwip_read 64
[20:19:08]! Unknown: GET /connecttest.txt
[20:19:08]lwip_read 47
[20:19:08]Unknown
[20:19:08]Redirecting to http://192.168.4.1/settings
[20:19:08]lwip_read 0
[20:19:08]Client disconnected
[20:19:09]lwip_read 64
[20:19:09]! Unknown: GET /connecttest.txt
[20:19:09]lwip_read 47
[20:19:09]Unknown
[20:19:09]Redirecting to http://192.168.4.1/settings
[20:19:09]lwip_read 0
[20:19:09]Client disconnected
[20:19:16]scandone
[20:19:21]station: e4:a7:a0:ac:7e:7a leave, AID = 1
[20:19:21]rm match
[20:19:26]scandone
[20:19:36]scandone
[20:19:46]scandone
[20:19:56]scandone
[20:20:06]scandone
[20:20:16]scandone
[20:20:26]scandone
[20:20:36]scandone
[20:20:46]scandone
[20:20:56]scandone

'''

i3laze, 2020-05-11:
GET and POST requests your client smarthphone/laptop makes to HAA portal are weird.
Try re-flashing the binary and/or some other gadget to setup HAA.

5high, 2020-05-12:
Thank you for you replay , i am give up using latest bin file, i am using old one working fine , Thanks.

closed 2020-06-11 by stale[bot]