RavenSystem/esp-homekit-devices #932: Not really a issue, but a wish

palletodap, 2020-05-09:
I really love this firmware, and have used it on like 10 sonoff products in my home, but also helped friends and family set it up for garage gates and so on.

I would really really like to have a way to remotely change the json config.

Would it be possible to make it possible to pull a json config from a url? 

I thought about this idea:
- Insert a URL in the json textbox in setup mode.
- Then every time the product reboots, it will pull the string from that url and use it instead.

This way i could set up custom configs from my web server, and change it even if i'm not there locally :)

Again, thanks for this amazing firmware! :D

tonysprenk, 2020-05-09:
I think this is a great idea. As a workaround maybe you can vpn into the network that the accessories are setup on and access them that way?

palletodap, 2020-05-09:
> I think this is a great idea. As a workaround maybe you can vpn into the network that the accessories are setup on and access them that way?

Yea thats an great idea also. I tried this some time ago, but then i need a guy close to put the product in setup mode before i could access :/

tonysprenk, 2020-05-09:
You can’t have access to the home in the home app?

tonysprenk, 2020-05-09:
In your first idea you would also need someone with access to the accessory in the home app or physically to reboot or put in setup mode. Better to have access to the home in the home app and then vpn into the network and then invoke setup mode.

palletodap, 2020-05-09:
> In your first idea you would also need someone with access to the accessory in the home app or physically to reboot or put in setup mode. Better to have access to the home in the home app and then vpn into the network and then invoke setup mode. 

Sure, that’s an option too. But if it’s hooked up to like an garage gate, then I do not want to open/close the gate 8 times remotely 😃

nikbyte, 2020-05-09:
I was also thinking about a way to enter setup remotely.
Maybe add http get request or port knocking for this as option?

tonysprenk, 2020-05-09:
Entering setup mode with a dummy switch is the easiest way.

palletodap, 2020-05-10:
> Entering setup mode with a dummy switch is the easiest way.

That’s right! Didn’t think about that. But i still think the pull from url would give a lot more flexibility 😋

nikbyte, 2020-05-10:
+1. Also extra switches would be visible by people who don’t need to see and switch them.

RavenSystem, 2020-05-10:
Sorry, but this will never happen. A secondary server listening for a request to enter setup mode is a big security hole and a it will add instability to the system.

closed 2020-05-10 by RavenSystem