RavenSystem/esp-homekit-devices #933: Orvibo S25US

richardlimaa, 2020-05-11:
Hi dear, is it possible to include this device?
https://domoticaencasa.es/tutorial-como-cambiar-el-firmware-a-un-orvibo-b25/
I'm no familiar, but it works with you firmware haaboot.bin, the only problem is that doesnt have the right pins, so relay cant work, but i did the configurations in arduino-1.8.12, so i could turn on and turn off with the button, but once did it, homekit stopped, this device use QW8266-02Q

i3laze, 2020-05-11:
HAA is intended to work will all esp82xx devices around.
You yourself set the right pins and their actions.

Check this short step-by-step guide and adapt to pins you mentioned above:
https://github.com/RavenSystem/esp-homekit-devices/issues/931#issuecomment-626382358

richardlimaa, 2020-05-11:
Hi my friend, i tried what you said, i setup a HAA Json Configurator but i dont know if its wrong, its appear in the homekit devices but wont work again. After apply the right pins on arduino-1.8.12 got same as before, i can start using pins but never in homekit app.

this is was i setup:
 
{"c":{"l":12,"b":[{"g":0,"t":5}],"i":1},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"t":2,"b":[{"g":14}],"s":0}]}

richardlimaa, 2020-05-11:
Hi my friend, you give a very good light! I got it i can turn on and turn off Orvibo S25US from button and home app, i use template from Sonoff S20/S26, so i change this pins in got, have sucess with this one here:

{"c":{"l":4,"b":[{"g":0,"t":5}],"i":0},"a":[{"0":{"r":[{"g":5}]},"1":{"r":[{"g":5,"v":1}]},"t":2,"b":

The only thing i didnt solve is the blue and red light, by the way its working verrrry good!

i3laze, 2020-05-12:
I would give you a hint about lights, but please copy your current JSON string correctly.

richardlimaa, 2020-05-12:
i3Laze, thanks for help me, and i tried you but only blue light i got! But i change some parameters and got the right colors, so blue when on, and red when off. This is the final:

{"c":{"b":[{"g":0,"t":5}],"i":0},"a":[{"0":{"r":[{"g":5},{"g":12,"v":1},{"g":4,"v":0}]},"1":{"r":[{"g":5,"v":1},{"g":4,"v":1},{"g":12,"v":0}]},"t":2,"b":[{"g":14}],"s":0}]}

\o/

Thanks so much, i guess this could be include in the database!

closed 2020-05-13 by richardlimaa