RavenSystem/esp-homekit-devices #936: Sensor initial state

i3laze, 2020-05-11:
I'm struggling with leakage sensor, that should perform action (close valve) on state change.
It is either possible to correctly read state on startup:
`{"0":{"r":[{"g":12,"v":1,"i":26}]},"t":7,"s":4,"g0":[{"g":9,"t":1}],"g1":[{"g":9,"t":0}]}`
or perform an action after state change, but startup "Sensor Status OFF" until sensor flaps:
`{"0":{"r":[{"g":12,"v":1,"i":26}]},"t":7,"s":4,"f0":[{"g":9,"t":1}],"g1":[{"g":9,"t":0}]}`

According to Wiki:
> The Initial State key is not supported by this accessory. Refer to Initial State for more details.

What's the difference again between f[n] and g[n]?
I need both action at Falling edge OFF and correct sensor state on startup.

i3laze, 2020-06-01:
Up.

i3laze, 2020-06-26:
I’d say Up again,

RavenSystem, 2020-07-05:
Initial sensor states are determined by GPIOs.

closed 2020-08-04 by stale[bot]