RavenSystem/esp-homekit-devices #937: Sonnof RF Bridge

mitchoklemen, 2020-05-11:
Hi,
Is there any chance to flash Sonoff RF Bridge with Homekit?
I mean ...
 https://sonoff.tech/product/accessories/433-rf-bridge

bj-tracer, 2020-05-12:
Hello!
Sonoff RF Bridge have an UART-controller. It convert RF-data from RF-receiver and send to esp8266. And back.
So you need to decode UART commands at the the esp for full working. But esp can only sending UART commands.

i3laze, 2020-05-21:
Sorry, author told million times RF433 is not on a roadmap.

Although you [can use pre-configured RF](https://github.com/RavenSystem/esp-homekit-devices/issues/689#issuecomment-619413650) models like Sonoff Basic R3 RF, that's it.

Alternative is you purchase native wall-mount RF433 switch or remote  and alter them with wires from Sonoff SV or Shelly.Cloud 2.5 (which support mouting in-wall behind standard wall switch)

i3laze, 2020-05-21:
UPD: Looks like Sonoff RF Bridge works pretty neat with Tasmota:
https://community.home-assistant.io/t/dooya-dt82tv-curtain-motors-with-sonoff-rf-bridge/97555

closed 2020-06-20 by stale[bot]