RavenSystem/esp-homekit-devices #939: Shelly 2.5 not show in wi-fi

josleng, 2020-05-11:
hello I have shelly 2.5. I have no original backup.
tried to flash with ravencore but don't show in my wi-fi device. how I can restore it with original firmware if I don't have a backup? why I can't see in wi-fi device?? thank you so much

peros550, 2020-05-12:
This is google’s first result on how to restore shelly’s firmware 

https://www.shelly-support.eu/forum/index.php?thread-amp/438-shelly-2-5-restoring-the-firmware/

I would suggest though to try a bit harder . HAA is an extremely good firmware and it works pretty well for the majority of the community in this repo.

i3laze, 2020-05-12:
How exactly did you try to flash it? Simply by uploading fullHAAboot.bin to Web GUI?

Got Shelly 2.5 working just fine as with double wall switch.

josleng, 2020-05-12:
Downloaded the original firmware. I have erased the shally with Mac terminal And put the bin file like restore a backup. But I can’t see in Wi-fi device. I tried to put again rave core but also I can’t see in WiFi.

tonysprenk, 2020-05-12:
@josleng probably best to just follow the installation instructions in the wiki. https://github.com/RavenSystem/esp-homekit-devices/wiki/Installation

closed 2020-06-18 by stale[bot]