RavenSystem/esp-homekit-devices #941: Setup sprinklers to irrigate one after one

morphix9, 2020-05-12:
Hey guys,

I've seen that many of you have set up sprinklers. I figured out how to set up timers, but I'm struggling how to get them together like this:

[image]

This appears to be possible with Rachio's implementation to Homekit. Has anyone figured this out? 

It's strange that sprinklers do not show up as accessories that could be added to scenes or automations. With the EVE app, you could add them to scenes, but why can't you do this with Home? Does HAA use a simpler integration? 

Thanks for your help!

i3laze, 2020-05-12:
Indeed, somehow currently we can’t create sprinkler scenes and automations in Home.app.
Similar discussion:
https://community.home-assistant.io/t/homekit-and-sprinklers/89978

But, you can do it and prevent create zones in 3rd party app (Rachio, Eve) and then use it in a home.app.

i3laze, 2020-05-12:
According to Rachio support site Zones are part of Sprinkler HomeKit accessory:
https://fast.wistia.com/embed/iframe/pujxdkut64?fbclid=IwAR1ErcdxGqkc7ohS7xv9tviT2QGriBJd4NqTU8w1bVPLAWYQS7OPv6bjJCY

As water pressure is limited, running multiple valves in serial (one by one) is common case.
@RavenSystem, any ideas how we can evolve in this direction? 
I suggest to add actions (to trigger remote HAA valve accessories) to Sprinkler accessory and represent them as Zones.

morphix9, 2020-05-13:
> Indeed, somehow currently we can’t create sprinkler/valve scenes and automations in Home.app.
> Similar discussion:
> https://community.home-assistant.io/t/homekit-and-sprinklers/89978
> 
> But all this is possible in 3rd party apps (Rachio, Eve) and then you trigger them in a Home.app.

I've tried Eve app, but can't figure out how to create zones. Can you use Rachio app without a Rachio HW and create zones? I really doubt that. 

There must be support for this natively. Ideally, each sprinkler added to a 4ch Sonoff would make a separate zone.

morphix9, 2020-05-13:
It looks like we need to create an "irrigation system" with the collection of valve accessories. From Apple's documentation:

```
A sprinkler system accessory may be:

• a combination of ”8.21 Irrigation System” (page 145) service on a bridge accessory with a collection of one or more ”8.43 Valve” (page 155) services (with ”9.124 Valve Type” (page 232) set to Irrigation) as bridged accessories (The bridge accessory is typically connected to each valve using wires). OR

• a combination of ”8.21 Irrigation System” (page 145) service with a collection of one or more linked ”8.43 Valve” (page 155) services (with ”9.124 Valve Type” (page 232) set to Irrigation) (The valves are collocated e.g. hose based system). OR

• a combination of ”8.43 Valve” (page 155) service(s) with ”9.124 Valve Type” (page 232) set to Irrigation (e.g., a system with one or more valves which does not support scheduling)
An irrigation system is set to Active when the system is enabled. When one of the valves is set to In Use, the irrigation system must be set to in use.

An accessory which includes this service must include the ”9.112 Set Duration” (page 223) in the ”8.43 Valve” (page 155).

An irrigation system accessory which does not auto detect the number of valves it is connected to and requires user to provide this information must include the ”9.49 Is Configured” (page 182) in the ”8.43 Valve” (page 155)
```

@RavenSystem would it be possible to enhance the "water valve" type?

i3laze, 2020-05-24:
As @RavenSystem replied around, there’s no plans for inter-HAA notifications.
Meaning, the valve system if developed, will be currently limited to Sonoff 4CH Pro and other multi-output units.

However, there might be a Tasmota sprinkler plugin for the cause of multi-device system.

closed 2020-06-23 by stale[bot]