RavenSystem/esp-homekit-devices #942: Control other physical devices

freddair, 2020-05-13:
Hi, 

I have two separate HK homes in one multi-party-house in a single network.

In one home I use a Shelly 1 as a door opener (lock mechanism) connected to the intercom.

I would like to have a dummy device in the other home (running on another HAA Shelly) that mirrors the lock mechanism so that I can use it to open the front door from both HK homes.

Is it possible to set this up using http commands or any other way?

I only found how to publish http commands but not how to control another HAA device with it.

Thanks a lot!

i3laze, 2020-05-13:
+1.
As I mentioned [here](https://github.com/RavenSystem/esp-homekit-devices/issues/941#issuecomment-627590688), there’s an urge for inter-HAA actions, which not rely on HomeKit Automations. 
Inside single HAA device there are already Notifications between accessories.

Custom HTTP requests with native HK API JSON queries like below are too complicated:
 `curl -X PUT -d '{"characteristics":[{"aid":'$aid',"iid":'$iid',"value":true}]}' http://$IP:5556`

peros550, 2020-05-13:
+1 from me on this clever request. 

Some automations although can be handled by Homekit hub, is not worth it to be handled through the hub. We are adding more and more HAA devices in our homes, and the need for automations is constantly increasing thus whatever app you are using to manage Home automations  gets cluttered. 

I am of the opinion to hardcode some very simple automations, that would never need any further tinkering on the esp8266 in a “set and forget it” fashion and leave space for more complex automations to be handled by the hub.

RavenSystem, 2020-05-19:
This is not possible for security and stability reasons.

closed 2020-05-19 by RavenSystem