RavenSystem/esp-homekit-devices #944: Functional click vs double click bug

nikbyte, 2020-05-13:
Hello,

```
{"c":{"l":13,"z":20,"b":[{"g":0,"t":5},{"g":4,"t":5}]},"a":[{"t":3, "f0": [{ "g": 4, "t": 0 }], "f1": [{ "g": 4, "t": 2 }],"f2": [{ "g": 4, "t": 3 }]}]}
```

With this JSON device generates f0 and then f1 on double click on latest firmware for today.
And on a second press it generates f0 and then f2.

On older firwares it worked well. Possible to fix?
Thanks.

nikbyte, 2020-05-14:
Any updates? On latest firmware double click and delayed click are broken.

RavenSystem, 2020-05-19:
Double click works perfectly. But your JSON is wrong. You are combining "t":0 with other press types, and combining LOW with HIGH states can have undesirable effects.

closed 2020-05-19 by RavenSystem

nikbyte, 2020-05-19:
Can you advice how to fix my json to work properly?

RavenSystem, 2020-05-19:
Change "t":0 to "t":1.