RavenSystem/esp-homekit-devices #945: Blitzwolf BW-LT11 and BW-LT21 issues

Kienz, 2020-05-14:
If I'm power off the lightstrip or the lightbulb it switches shortly to red (show red color) before it is powered off. And the color color accuracy is for some colors really bad compared to a eg. a Philips Hue lightstrip - to get the color orange you have almost set to red.

JSON:
BW-LT11
```{"a":[{"t":30,"r":4,"g":12,"v":14,"w":5},{"1":{"s":[{"a":1}]}},{"1":{"s":[{"a":2}]}}]}```
BW-LT21
```{"a":[{"t":30,"r":14,"g":12,"v":5,"w":15},{"1":{"s":[{"a":1}]}},{"1":{"s":[{"a":2}]}}]}```

Does anybody has the same problems? Any help?

nikbyte, 2020-05-14:
+1. I have the same issue. Who knows the secret?

peros550, 2020-05-14:
Have faced the same issues with Blitzwolf LT11 and LT21 in trying to get red and orange colors. It's like the bulb does not have enough shades around orange and red, thus it gives different color when trying to play around orange or red. For example when trying to set red, I get a lot of pink tones until I get a true red color.

closed 2020-06-13 by stale[bot]