RavenSystem/esp-homekit-devices #948: External toggle problem

tmps99308091, 2020-05-14:
How can I set up an json string for an external toggle which turn on or off a switch when the state of an external toggle changes.

Kienz, 2020-05-18:
Not sure what you mean. Can you explain it better please (eg. circuit diagram, etc.)

tmps99308091, 2020-05-18:
I've figure it out. The json string is "b":[{"g":1,"t":1},{"g":1,"t":0}].
I want to control the light at home but the original dummy switch should be able to control the light too. This json string enables the external toggle to be toggle when the external pin gets from low to high or from high to low.

bj-tracer, 2020-05-21:
Hi!
Go to Examples at the Wiki ;))))
https://github.com/RavenSystem/esp-homekit-devices/wiki/HAA-Templates
The first template has answer for your question.
`"b":[{"g":0,"p":1,"t":1},{"g":14,"p":1,**"t":1**},{"g":14**,"t":0**}]`
You need to use 2 opposite types of input for one GPIO.

closed 2020-05-21 by tmps99308091