RavenSystem/esp-homekit-devices #950: Sonoff Mini

eminencegrs, 2020-05-15:
Hi!

Please, could you tell me this firmware supports Sonoff Mini DIY?

nikbyte, 2020-05-15:
Yes. I use it with Sonoff Mini.

shanghaix, 2020-05-15:
Yes, you don’t have to solder anything, just flash it OTA with the DIY tool and put Tasmota lite on it, then from Tasmota you can upgrade to HAA with the firmware provided in the Wiki.

bj-tracer, 2020-05-21:
> Yes, you don’t have to solder anything, just flash it OTA with the Sonoff DIY tool and put Tasmota lite on it first, then from Tasmota you can upgrade to HAA with the firmware provided in the Wiki https://github.com/RavenSystem/esp-homekit-devices/wiki/Installation#installing-from-tasmota
> 
> Not sure if you can directly upload HAA without having to go through Tasmota first, haven’t tried.

Your suggestion is for flashing for HAA firmware.
But sonoff mini at DIY mode can be controll without this.

bj-tracer, 2020-05-21:
> Hi!
> 
> Please, could you tell me this firmware supports Sonoff Mini DIY?

Привет! :)
Если есть любое устройство прошитое HAA, то можно в нем создать виртуальный аксессуар, который будет посылать HTTP-команды на Sonoff mini. Команды описаны в DIY API на сайте Itead.

(Hi!
You can add virtual accessory for any HAA-device. It will be controll the Sonoff mini by HTTP-request. You can read about commands at the Itead Wiki web-page.
)

nikbyte, 2020-05-21:
Why do you need that if you can just flash HAA into Sonoff MINI directly?

bj-tracer, 2020-05-22:
> Why do you need that if you can just flash HAA into Sonoff MINI directly?

Cause it's not a DIY mode ;))) It's flashing for new firmware.

eminencegrs, 2020-05-28:
Thanks guys!

closed 2020-06-27 by stale[bot]