RavenSystem/esp-homekit-devices #951: Unable to update WiFi SSID after first setup.

seshaln, 2020-05-15:
Sonoff R4 (ESP8285)
Steps to reproduce:
1.Install HAA according to manual https://github.com/RavenSystem/esp-homekit-devices/wiki/Installation
2. Connect to HAA-xxxxxx Wi-Fi, select my SSID MY_NETWORK, enter a password.
3. Wait for OTA update(about 5 minutes)
4. Sonoff rebooted, and successfully connected to MY_NETWORK. HomeKit works OK.
5. Reboot Sonoff R4 in setup mode. Connect to HAA-xxxxxx, select new SSID SSID MY_NETWORK_NEW, enter a password.
6. FAIL - Sonoff R4 still trying to connect to old network (MY_NETWORK), and showing it in UART logs.
I found only one way to set up new SSID - reflash Sonoff R4 form start.

i3laze, 2020-05-16:
As usually simple SSID select is working, in your case it might be needed to use Reset WiFi option - it is available in the same Setup mode.

seshaln, 2020-05-17:
Of course, I checked `Reset WiFi` option - no success 😞

RavenSystem, 2020-05-20:
Use last version.

closed 2020-05-20 by RavenSystem