RavenSystem/esp-homekit-devices #952: Use multiple Shellies on my homekit

VanAxl, 2020-05-15:
Hi Everyone, 
Well I'm wondering if I miss something on the installation guide, because I did successfully install one of my Shelly 1 device (garage door). But I have a second one I would like to use on my homekit installation, I've setup this second as a simple switch, but when I try to scan the homekit code, it says that this device is already registered !

So if someone could tell me how to add multiple shells on my homekit ? 
Would be nice.

Thanks a lot.
VanAxl.

Kienz, 2020-05-15:
Enter the code 02182017 manually. This works - sometimes on the 2nd attempt.
I have about 20 devices from Shelly and all ca be added.

VanAxl, 2020-05-15:
Great, you are right, it was that simple, manually adding the code ! I was always scanning .... thanks a lot.

closed 2020-05-15 by VanAxl