RavenSystem/esp-homekit-devices #953: Possible to OTA flash off HAA?

pashdown, 2020-05-15:
Is it possible to OTA flash off HAA to Tasmota or Espurna?  I have some Gosund WP6's and they're impossible to serial flash without severely damaging the case.  They epoxied them shut.  I'd like to try some other firmwares, but now I'm on HAA, I noticed that it will only flash signed firmwares.

peros550, 2020-05-16:
+1 Going from Tasmota to HAA is possible. I believe it should be the opposite as well.

HomeACcessoryKid, 2020-05-17:
HAA-OTA is a stripped down clone from LCM which is different in three ways:
- it can only support HAA whereas LCM supports any possible binary and the user code owner doesn't need a secret key to sign this code since GitHub will protect the authenticity.
- because it only works for HAA, HAA-OTA dropped the certificate check on GitHub and you can get away with a spoofed server.
- instead @RavenSystem signed his binaries which he can do because he only has to support HAA

LCM versions 2.0.0 and higher solved the block-out dilemma by making an emergency mode where a new signed OTABOOT file can be loaded from a local server. So depending if you want to ever switch away from HAA you can use HAA-OTA or LCM instead.

So, the answer is that you should use LCM if ever you want to use alternative software on your device.

peros550, 2020-05-17:
@HomeACcessoryKid many thanks for everything you 've done and keep doing in this community. 
You answered to the question I didn't made :) but I was always had in my mind. 

Problem is that going with LCM, there is a small possibility that Ravensystem cannot support it anymore in case HAA firmware gets bigger.

nikbyte, 2020-05-17:
@HomeACcessoryKid is there any way to migrate from latest HAA_OTA to latest LCM?
I see LCM in https://github.com/RavenSystem/haa/releases repository, however it's not signed with .sec file as signed otamain.bin . It looks like without @RavenSystem signature I can't migrate to LCM. Or is there any way?

nikbyte, 2020-05-17:
Maybe @RavenSystem could sign haa_lcm.bin with his key too for this?

HomeACcessoryKid, 2020-05-17:
that version haa_lcm.bin exists because it uses a different sysparam setting than haa main.bin. It was made so it matches the LCM and general esp-open-rtos values for sysparam.
It does not even try to do any OTA tasks so it cannot be used for any transfer tasks or so.

For HAA-OTA to transfer to LCM, @RavenSystem would have to make code that
1. makes it possible for someone to indicate they want this
2. to then actually load the otaboot.bin from LCM
3. to make sure it still matches the LCM conventions of sysparam values

nikbyte, 2020-05-17:
@RavenSystem made support of own servers. So, #1 can be achieved by placing LCM OTA there. However currently this isn't possible because there is no signature from @RavenSystem for LCM OTA binary.

HomeACcessoryKid, 2020-05-17:
That makes sense to me. But it would have to be  the LCM otaboot.bin file that gets signed. He can collect a copy from GitHub and sign it in the required way.

nikbyte, 2020-05-17:
@RavenSystem could you please sign for us a copy of otaboot.bin from LCM?

RavenSystem, 2020-05-17:
Here is. Remember that haa_lcm will drop support soon to gain flash storage for next versions.

[otaboot_signed.zip](https://github.com/RavenSystem/esp-homekit-devices/files/4640430/otaboot_signed.zip)

nikbyte, 2020-05-17:
Thank you. Let me try.

nikbyte, 2020-05-17:
@RavenSystem could you please explain a bit more your words?
> Remember that haa_lcm will drop support soon to gain flash storage for next versions.
You mean that HAA will not work with LCM properly some time later?

pashdown, 2020-05-17:
Hi. Maybe we got off track here, but after flashing otaboot.bin, I'm stuck in some sort of flashing loop.

`nc -kulnw0 45678
OK SSL..OK set_fd to github.com port 443..OK
sent OK
HTTP returns 302
--- ota_connect LocalPort=f5fb DNS IP:52.216.78.68 local..OK remote..OK SSL..OK set_fd to github-production-release-asset-2e65be.s3.amazonaws.com port 443..eras--- ota_connect LocalPort=f5fc DNS IP:52.216.78.68 local..OK remote..OK SSL..OK set_fd to github-production-release-asset-2e65be.s3.amazonaws.com port 443..OK
--- ota_connect LocalPort=f5fd DNS IP:52.216.78.68 local..OK remote..OK SSL..OK set_fd to github-production-release-asset-2e65be.s3.amazonaws.com port 443..eras--- ota_connect LocalPort=f5fe DNS IP:52.216.169.99 local..OK remote..OK SSL..OK set_fd to github-production-release-asset-2e65be.s3.amazonaws.com port 443..era--- ota_connect LocalPort=f5ff DNS IP:52.216.169.99 local..OK remote..OK SSL..OK set_fd to github-production-release-asset-2e65be.s3.amazonaws.com port 443..OK
erasing@0x41000> collected 261312 bytesterasing@0x40000> collected 258048 bytes
--- ota_verify_hash
--- ota_hash
--- ota_finalize_file
--- ota_write_status
--- ota_reboot

--- user_init
--- OTAMAIN VERSION: 2.0.2
--- pre_wifi_config
--- ota_read_rtc
--- count_step=3
--- count=1
--- standard ota
--- on_wifi_ready
--- ota_emergency?
--- ota_init
userbeta='0' otabeta='0'
rboot_config.unused[1]=LEDinfo from 0x00 to 0x00
--- ota_active_sector: 0xfa000
--- ota_set_verify...OFF
--- DNS: done!
--- ota_get_pubkey
ret: 0
--- ota_get_btl_version
bootloader version:"1.0.1"
--- ota_boot...1
--- ota_load_user_app
user_repo='HomeACcessoryKid/ota-demo' user_version='0.0.0' user_file='main.bin'
--- entering the loop
--- ota_get_pubkey
ret: 0
--- ota_set_verify...OFF
--- ota_get_version
--- ota_connect LocalPort=e0b3 DNS IP:140.82.112.4 local..OK remote..OK SSL..OK set_fd to github.com port 443..OK
GET /HomeACcessoryKid/life-cycle-manager/releases/latest HTTP/1.1
Host: github.com

HTTP returns 302 for --- ota_boot...1
HomeACcessoryKid/life-cycle-manager@version:"2.0.2"
--- ota_get_hash
--- ota_get_file_ex
GET /HomeACcessoryKid/life-cycle-manager/releases/download/2.0.2/certs.sector.sig HTTP/1.1
Host: github.com

--- ota_connect LocalPort=e0b4 DNSOK SSL..OK set_fd to github.com port 443..OK
sent OK
HTTP returns 302
--- ota_connect LocalPort=e0b5 DNS IP:52.216.241.236 local..OK remote..OK SSL..OK set_fd to github-production-release-asset-2e65be.s3.amazonaws.com port 443..OK
certs size: 2628
TIME: Sun May 17 20:15:37 2020
--- ota_get_version
--- ota_connect LocalPort=e0b6 DNS IP:140.82.112.4 local..OK remote..OK SSL..OK set_fd to github.com port 443..OK
GET /HomeACcessoryKid/life-cycle-manager/releases/latest HTTP/1.1
Host: github.com

HTTP returns 302 for --- ota_boot...1
HomeACcessoryKid/life-cycle-manager@version:"2.0.2"
--- ota_get_hash
--- ota_get_file_ex
GET /HomeACcessoryKid/life-cycle-manager/releases/download/2.0.2/certs.sector.sig HTTP/1.1
Host: github.com
[... 686 lines cut ...]
HTTP returns 302 for --- ota_boot...1
HomeACcessoryKid/life-cycle-manager@version:"2.0.2"
--- ota_get_hash
--- ota_get_file_ex
GET /HomeACcessoryKid/life-cycle-manager/releases/download/2.0.2/certs.sector.sig HTTP/1.1
Host: github.com

--- ota_connect LocalPort=c6a1 DNS IP:140.82.114.3 local..OK remote..OK SSL..OK set_fd to github.com port 443..OK
sent OK
HTTP returns 302
--- ota_connect LocalPort=c6a2 DNS IP:52.216.163.11 local..OK remote..OK SSL..OK set_fd to github-production-release-asset-2e65be.s3.amazonaws.com port 443..OK
 collected 155 bytes
--- ota_boot...1
--- running ota-main software
--- ota_get_version
--- ota_connect LocalPort=c6a3 DNS IP:140.82.114.3 local..OK remote..OK SSL..OK set_fd to github.com port 443..OK
GET /HomeACcessoryKid/rboot4lcm/releases/latest HTTP/1.1
Host: github.com

HTTP returns 302 for --- ota_boot...1
HomeACcessoryKid/rboot4lcm@version:"1.0.1"
--- ota_compare 1.0.1 with 1.0.1=0
--- ota_compare 2.0.2 with 2.0.2=0
--- ota_get_version
--- ota_connect LocalPort=c6a4 DNS IP:140.82.114.3 local..OK remote..OK SSL..OK `

nikbyte, 2020-05-17:
+1, the same loop.

HomeACcessoryKid, 2020-05-17:
OK, if you read above you saw that there was a bullet number 3
- to make sure it still matches the LCM conventions of sysparam values

`user_repo='HomeACcessoryKid/ota-demo'`
You are loading the repo ota-demo which has as its purpose to request another update after 30 seconds without user input.

I do not know if your repo before was haa, but it seems it was not compatibly stored before, violating condition 3.  I would need full log, not just the second part of it...

BUT, read the instructions of LCM and use the power cycles to get into a full factory reset (15 cycles) and then you should be able to set wifi, and your own repo...

And then think about the reason this did not work out of the box...

nikbyte, 2020-05-17:
@HomeACcessoryKid I set own repo, it searches latest version and then go to life-cycle-manager repo.

HomeACcessoryKid, 2020-05-17:
please report on results of the factory reset?

nikbyte, 2020-05-17:
@HomeACcessoryKid I'm trying, but can't initiate the factory reset yet.

Also I have a question. If I'm on LCM, is there a way to migrate back to HAA?

nikbyte, 2020-05-18:
@RavenSystem Could you please email me <login> @ gmail.com ? I have a question to you for donation.

HomeACcessoryKid, 2020-05-18:
You can use HAA by specifying it as the repository. You cannot revert (yet) back to HAA-OTA, but using the same trick as switching from HAA_OTA to LCM, you could switch back as well.

nikbyte, 2020-05-18:
@HomeACcessoryKid But my flashing from the latest HAA_OTA to LCM is unsuccessful yet. :( Also I can't reset to factory defaults. And now I have just infinite loop.

HomeACcessoryKid, 2020-05-18:
@nikbyte You indicated to have the same loop as @pashdown and that clearly shows the use of LCM: `--- OTAMAIN VERSION: 2.0.2`

when doing power cycles, try to do the first one at about 1s and any subsequent can be within 4s per cycle and use `nc -kulnw0 45678`

The log will show you how often the boot loader counted for you. This will help you adjust the cycling if needed.

nikbyte, 2020-05-18:
@HomeACcessoryKid but you understand what is the reason of that loop?

HomeACcessoryKid, 2020-05-18:
of course, it is the otademo https://github.com/HomeACcessoryKid/ota-demo/blob/master/main.c which automatically triggers a new ota cycle every 30 sec.
It looks as if this 'side door' of switching from HAA-OTA to LCM selected the default repo.
It is a use case that normally would not occur.

nikbyte, 2020-05-18:
I placed HAA (not OTA) to my own repository on github and set this as repository. But I see one try to check version, that version is not equal and then it go to LCM repository and go into into infinity loop. I don't use ota-demo. Let me generate log and how you.

nikbyte, 2020-05-18:
[1.log](https://github.com/RavenSystem/esp-homekit-devices/files/4646456/1.log)

HomeACcessoryKid, 2020-05-18:
in order to prevent download errors you must make a main.bin.sig file as stated in the [LCM instructions](https://github.com/HomeACcessoryKid/life-cycle-manager#creating-a-user-app-digitalsignature)
In the log you can see:
```
GET /nikbyte/test/releases/download/2.3.3/main.bin.sig HTTP/1.1
Host: github.com

--- ota_connect LocalPort=dcd7 DNS IP:52.64.108.95 local..OK remote..OK SSL..OK set_fd to github.com port 443..OK
sent OK
HTTP returns 404
--- ota_reboot
```
And because LCM takes care that if things do not work the way they were intended, it will try again, and again, etc... Just so you do not get a bricked device if finally you fix the underlying problem.

nikbyte, 2020-05-18:
@HomeACcessoryKid, it works! thanks a lot.

HomeACcessoryKid, 2020-05-18:
enjoy

closed 2020-06-18 by stale[bot]

jakobsudau, 2020-10-10:
Hello!
I know this issue is already closed, but can someone help me flash my HAA device back to Tasmota? Some step-by-step instructions would be very much appreciated! I am a little confused following this conversation and I am no expert.

The device is running HAA v3.5.7 and HAA OTA v4.2.5

1. Download the otaboot_signed.zip from RavenSystem earlier in this conversation.
2. Setup a local server and serve the otaboot_signed.zip file/firmware.
3. Enter Setup Mode of device.
4. In the Setup Mode of the device, down in "Custom server for updates:" choose the according URL and corresponding port.
5. In the Setup Mode of the device, click on "Update all now" to start the update process loading the otaboot_signed.zip firmware.

How do I proceed then? Where do I specify the Tasmota firmware URL or file?

Thanks for your feedback!

healeydave, 2021-01-27:
> Hello!
> I know this issue is already closed, but can someone help me flash my HAA device back to Tasmota? Some step-by-step instructions would be very much appreciated! I am a little confused [...]

Did you ever figure this out?

I am in the same boat, in fact I didn't even intend to put HAA onto my device, I picked the wrong firmware by accident.
I have never come across a system before where-by their is no easy way back, very frustrating!!