RavenSystem/esp-homekit-devices #957: Why bloat HAA with ALL features and make it impossible to keep things simple

HomeACcessoryKid, 2020-05-17:
In one way I understand the ease of having only one binary and never look back when you grow and grow. But at the same time HAA software serves a narrow goal.
At any given moment when it gets loaded in a particular device like a sonoff basic switch it has to do its one thing: switch and light a led. not more. And more importantly, it will not suddenly grow an IR sender.

I hear that HAA will grow even more and that is great for those very special devices that need those features. But the majority of simple devices couldn't care less and all they want is security updates if they would exist.

This is a plea to @RavenSystem to put in some sense to the communities needs.
It will take a  bit of extra effort but will make a lot more sense.

Separate the HAA code in 2, 3 or 4 pieces, which you can enable at compile time. This would give us 2-4 binaries that are able to work for each category.

The HAA-basic.bin would serve for 75% of the cases and is probably already fully developed.
It can be compatible with LCM based ota software and - most importantly - doesn't have to get an update as often since this code is very stable already. Also the risk of something basic breaking down because of some new feature is reduced.

The HAA-IR.bin would have features that support IR, not needed in a lot of devices.

etc.

So, what does the rest of you say?

RavenSystem, 2020-05-19:
Most of people uses HAA with HAA OTA, that allows to use a local server for updates, and allows to downgrade versions, letting me to create a bigger binary. I am not interested on create a lot of different versions, with IR, TCP/IP requests, pings, accessory types, etc. The main idea is to keep HAA simple to be maintained in an easy way.

HomeACcessoryKid, 2020-05-19:
The proposal was primarily to make 2 versions, but if that is not your plan, then I propose that I and/or other volunteers fork off your HAA and trim it down to make one HAA-basic

Not tomorrow, but if it becomes unfeasible to maintain HAA compatible with LCM.

If this happens, people would have to switch from HAA-OTA to LCM by using the system engineered in #953 and LCM would then make the switch under water.

morphix9, 2020-05-21:
> In one way I understand the ease of having only one binary and never look back when you grow and grow. But at the same time HAA software serves a narrow goal.
> At any given moment when it gets loaded [...]

I don't mind about a bigger image and support for IR (I don't use it), what I would hope for in general is:
- having beta and stable releases (being stable after they are really tested out and are bulletproof),
- adding support for full HomeKit repertoire (Sprinkler zones, security accessories, etc.)
- ability to watch HAA logs over the air (like for OTA boot) to report bugs/problems easier.

i3laze, 2020-06-02:
@HomeACcessoryKid, What’s the purpose again of having a separate branch? 
Are we running out of flash space?

HomeACcessoryKid, 2020-06-02:
RavenSystem is running out of space since he uses a “one binary for anything you can imagine”. But for 90% of the devices a basic code covers al the hardware can support anyway. That basic code could be small enough to be updated by lcm over the air. 

This would allow to us other code than haa if so desired in a later moment of its lifecycle 

BR,
HacK

closed 2020-07-02 by stale[bot]