RavenSystem/esp-homekit-devices #958: Trouble entering setup mode

shimon88, 2020-05-18:
Hi all
I’m having troubles entering into setup mode for some devices in different kind of versions. Either with emergency setup mode or via 8 time toggle, the device will create the SSID “HAA-xxxxxx”, but I can’t access the web UI to config. Not by navigate to the device ip, not by navigate to device host name and if I select the created SSID I get a message that I can’t connect. Either way after few seconds the device return to be available in Home. 

What am I doing wrong?

Thank you

shimon88, 2020-05-22:
Hi,
can any one help?

sorriso93, 2020-05-22:
same problem for me, I posted a log with example configuration

sorriso93, 2020-05-25:
Did you try to connect via the network (i.e. 192.168.0.x or with the local host name HAA-123456) or directly to HAA wifi? Once you setup the first time you should use the first method, not the second one. Take a look at the wiki, now it is more clear

tarontop, 2020-05-26:
> Did you try to connect via the network (i.e. 192.168.0.x or with the local host name HAA-123456) or directly to HAA wifi? Once you setup the first time you should use the first method, not the second one. Take a look at the wiki, now it is more clear

Unable to connect to WiFi, prompt unable to join the network

closed 2020-06-25 by stale[bot]