RavenSystem/esp-homekit-devices #959: Feature Request: support for water level sensor

tonysprenk, 2020-05-18:
It would be great to have support for the[JSN-SR04T](https://www.elecrow.com/water-proof-integrated-ultrasonic-ranging-module-jsnsr04t-p-1151.html) water level sensor. It is a supported sensor in [homekit](https://developer.apple.com/documentation/homekit/hmcharacteristictypewaterlevel).

i3laze, 2020-05-20:
Looks like range sensor to me..)
Seems it will require some offset parameters like <20cm = 100%, >1m=0%

tonysprenk, 2020-05-20:
> Totally doesn’t look waterproof :)
People online that have used it for the same purpose are saying it has worked for a long time.

wayn3, 2020-05-25:
this would be great

i3laze, 2020-06-06:
Similar JSN-SR04T device here with sensor on a  wire:
https://a.aliexpress.ru/_eNRfbf

Will it work with HAA?

I need to check garage occupancy before opening it in vain.

i3laze, 2020-06-06:
Same JSN-SR04T device here with sensor on a wire:
https://a.aliexpress.ru/_eNRfbf

@RavenSystem, please the check item description for 5 supported modes of operation:
Will it work with HAA?

I need to check garage occupancy before opening it in vain.

tonysprenk, 2020-07-03:
.

L4RGER, 2020-07-03:
> > Totally doesn’t look waterproof :)
> 
> People online that have used it for the same purpose are saying it has worked for a long time.

Yeh I thought the waterproof one was [This one](https://core-electronics.com.au/weatherproof-ultrasonic-sensor-with-separate-probe.html?utm_source=google_shopping&gclid=CjwKCAjwrvv3BRAJEiwAhwOdMw2kpzV867fJkcwNfOQMby-l2bgw-NTMAzDnIN36Q--myyy5VOuH7xoCcD0QAvD_BwE) with the long lead

tonysprenk, 2020-07-03:
Yeah I was think of this one: https://a.aliexpress.com/_dTQYgCg

RavenSystem, 2020-07-05:
Water level is an optional characteristic of Humidifier Dehumidifier service. Then, it is not an accessory by itself.

closed 2020-07-05 by RavenSystem

L4RGER, 2020-07-26:
@tonysprenk what do you mean when you say it is a supported sensor in homekit. 
I’m still figuring out mine and working out enclosures etc but I’ve got them sitting in HA ready. Just they aren’t supported through the homekit integration

tonysprenk, 2020-07-26:
Like @RavenSystem said it's only supported in HomeKit as part of a humidifier. My idea was to have a sort of reverse dimmer. So a dimmer in the home app but instead of using the slider to dim a light the slider will go up and down automatically as the water level falls and rises. I think technically this should be possible in JSON.

pilot1981, 2022-12-11:
I used a JSN SR04T with water leak sensor HAA accessory and a fixed distance: it works fine!