RavenSystem/esp-homekit-devices #960: Need some advice, I bought Apple Watch now all HAA devices broken

gerarddvb, 2020-05-18:
I bought an Apple Watch this week and now all HAA devices show 'no response' 
It isn't only HAA, homebridge devices also show 'no response'

I have tried rebooting devices, routers and even reinstalled everything.
After reinstall it works for like 5 minutes.

I created new house and add reinstalled devices, still it works for 5 minutes.
I don't have a clue what to do next. Anybody else have some advice?

MikeDProducer, 2020-05-18:
What are you using for a hub? Apple TV? HomePod? Try power cycling that.

gerarddvb, 2020-05-19:
Using AppleTV’s rebooted them all. I have now deleted entire house from homekit. Starting new may help

Kristian8606, 2020-05-19:
There seems to be a problem with Cloud.  it happened to me too.  After about a day, everything is fine.  If you are not waiting for a restart of Apple Tv should solve the problem.

gerarddvb, 2020-05-19:
Ok I will wait a day..

Things i tried today
- removed appletv’s from homekit
- reset all network settings on iphone
- rebooted all routers again
- force reboot All devices
- logged out and in icloud
- created new appleid and new house

Still works 5 minutes, going insane atm

Kristian8606, 2020-05-19:
Try restoring or reset your Apple TV

Kristian8606, 2020-05-19:
You can try to remove Apple tv from homekit then wait an hour for it to be reflected in the cloud then add all devices to homekit without apple tv.  Finally after you are sure that the devices are working add apple tv.  Remote mode may not work at first, but it works for me after about a day.

i3laze, 2020-05-19:
Everything should work with and without Watch and AppleTV.
Can you actually ping the devices? It might be Wifi issues and some coinsidence.

P.S. Raising wrist and asking for gates to open - priceless.

gerarddvb, 2020-05-19:
> You can try to remove Apple tv from homekit then wait an hour for it to be reflected in the cloud then add all devices to homekit without apple tv. Finally after you are sure that the devices are working add apple tv. Remote mode may not work at first, but it works for me after about a day.

I removed all appletv’s from homekit, it does not make any difference. I will restore the appletv’s and other i-devices this week. Have no time to do this now. Will let you know how it plays out.

gerarddvb, 2020-05-19:
> Everything should work with and without Watch and AppleTV.
> Can you actually ping the devices? It might be Wifi issues and some coinsidence.
> Check your WiFi router to find HAA-device IP addresses and verify they are currently connected to WiFi.
> 
> P.S. Raising wrist and asking for gates to open = priceless.

All devices ping without problems. Starting to think my icloud is fubar. I will restore every idevice soon, hope that helps.

HomeACcessoryKid, 2020-05-19:
I am investigating the unresponsive issue in general in the esp-homekit library
My current analyses says that if too many clients connect, then memory runs low and mdns cannot respond anymore. So, could you check if you can switch one or two clients that you use less into airplane mode and if things stabilise? That should be a proof for my theory.

curious, HacK

peros550, 2020-05-19:
Hey, roughly what’s the number of clients that bring HomeKit Into unresponsive state ?

HomeACcessoryKid, 2020-05-19:
it all depends on the accessory definition of the device, since that will consume a baseline of memory

We also defined a patch that removes an existing connection from a phone/ipad when a new one arrives. That still has to become a PR, but it does help.
Anyway, I have 7 distinct client devices but with the iOS that makes it possible for two connections per device, I definitely run into trouble. With the patch it is a lot better. This is for a security system and 6 motion sensor setup and that makes it 7 accessories on one esp8266. But on a leaner setup it might take 10 client devices or 15...

RavenSystem, 2020-05-19:
HAA does not use original esp-homekit library. It has systems to avoid duplicated connections and to keep memory usage and fragmentation in low levels. When I saw that kind of issue, there was a mDNS issue with the router or any other network device that drops all mDNS subsystem sending a mDNS flush flag.

@gerarddvb I suggest that you do a test using your iPhone sharing Internet as a WiFi router, configured with same WiFi settings as your router, turn off your router, and check if devices work connected to your iPhone. Or if you have other router, try it.

gerarddvb, 2020-05-20:
Thanks for all the replies!
It indeed was my ISP router.
Restored the router and HAA and homebridge are working for a couple of hours.
Hope it will stay this way. Need to see what happens in the next days.

closed 2020-05-20 by gerarddvb

gerarddvb, 2020-05-20:
> I am investigating the unresponsive issue in general in the esp-homekit library
> My current analyses says that if too many clients connect, then memory runs low and mdns cannot respond anymore. So, [...]

I have 8 esp devices connected to HK;
3 Shelly (switches)
2 Sonoff (outlets)
2 Wemos (temperature, doorbell)
1 ESP7 (temperature)

With that 3 homebridge plugins, 20 certified hue devices and some certified fibaro sensors. The certified devices had no issues during my connection problems.

I solved it with a router reinstall.
I use an ISP router and 3 Airports.
I only needed to reset the ISP router to factory defaults.

Hope this can help for your investigation, if you need more info let me know

peros550, 2020-05-21:
Perhaps it would be also helpful to list the ISP router models that give us headache. 
I for one, have been very happy with ASUS routers running Merlin firmware. On the other hand, I have faced some kind of issues with TP-Link routers.

gerarddvb, 2020-05-21:
Using Technicolor TC7210