RavenSystem/esp-homekit-devices #961: Accessory Already Added on second phone

hardheid, 2020-05-18:
I have flashed a Sonoff Basic R2 with haa version 2.3.3. It works, except that I can only add one device. The first device can ad the accessory, but on the second iOS device the message apprears 'Accessory Already Added'. 

I have tried reflashing, changing configs, changing the hostname, resetting the AppleID, changing the ip address, resetting iOS devices, but no luck. Some people seem to be facing the same issue on Homebridge. Any idea?

My config:
{
  "c": {
    "l": 13,
    "n": "sonoff-laptop",
    "o": 1,
    "u": 1,
    "ttl": 5,
    "b": [
      {
        "g": 0,
        "t": 5
      }
    ]
  },
  "a": [
    {
      "0": {
        "r": [
          {
            "g": 12
          }
        ]
      },
      "1": {
        "r": [
          {
            "g": 12,
            "v": 1
          }
        ]
      },
      "b": [
        {
          "g": 0
        }
      ],
      "s": 1
    }
  ]
}

tonysprenk, 2020-05-18:
You shouldn’t have to add an accessory to multiple devices. Once you add the accessory on one device iCloud takes care of the rest. But than both devices need to be logged in to the same Apple ID. If this is not the case you need to invite the second Apple ID to join the home in the home app on the first device.

Kienz, 2020-05-18:
Did you add the devices with the qr-code? Please try by manually enter the code 02182017.

hardheid, 2020-05-18:
> You shouldn’t have to add an accessory to multiple devices. Once you add the accessory on one device iCloud takes care of the rest. But than both devices need to be logged in to the same Apple ID. [...]

That works! Yes, different users. I didn't know that a Home could not be managed separately within HomeKit. 

> Did you add the devices with the qr-code? Please try by manually enter the code 02182017.

That doesn't work- "Can't add accessory"

closed 2020-05-18 by hardheid