RavenSystem/esp-homekit-devices #962: HAA Update went wrong

i3laze, 2020-05-19:
I've started update on several Shelly.Cloud units running HAA 2.3.1 and OTA 3.0.1 and it took too long. I've reset WiFi router a few times and eventually power.

Now they don't boot and don't go to emergyncy Setup mode/broadcast HAA-SSID:

I thought OTA code was resilient and continue flashing after reset. How so?

```
pp_task_hdl : 3ffef9a8, prio:14, stack:512
pm_task_hdl : 3ffef410, prio:1, stack:176
frc2_timer_task_hdl:0x3fff4b48, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ May 17 2020 18:10:18
phy ver: 273, pp ver: 8.3

Sysparam OK

WiFi Init
Fatal exception (28): 
epc1=0x40218827
epc2=0x00000000
epc3=0x00000000
excvaddr=0x00000005
depc=0x00000000
excsave1=0x40218824
Registers:
a0 40218824 a1 3fff38d0 a2  00000001 a3  00000000
a4  00000001 a5  00000000 a6  3fff3884 a7  3feffe00
a8  00000000 a9  0000001a a10 00000018 a11 04000102
a12 00000006 a13 3fff2510 SAR 00000008

Stack: SP=0x3fff38d0
0x3fff38d0: 3fff3910 00003002 0000ffff 00000001
0x3fff38e0: 69466957 00000000 00000000 00000000
0x3fff38f0: 00000000 00000000 00000000 00000000
0x3fff3900: 7274616d 32697661 00003233 00000000
0x3fff3910: 00000000 00000000 00000000 00000000
0x3fff3920: 00000000 00000000 00000000 00000000
0x3fff3930: 00000000 00000000 00000000 00000000
0x3fff3940: 00000000 00000000 3fff27dc 00000000

Free Heap: 45936
_heap_start 0x3fff29c8 brk 0x3fff4c90 supervisor sp 0x40000000 sp-brk 45936 bytes
arena (total_size) 8904 fordblks (free_size) 0 uordblocks (used_size) 8904

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 2292, room 16 
tail 4
chksum 0x57
load 0x3ffe8000, len 772, room 4 
tail 0
chksum 0x0b
csum 0x0b

rBoot v1.4.0 - richardaburton@gmail.com
Flash Size:   8 Mbit
Flash Mode:   DOUT
Flash Speed:  40 MHz
rBoot Option: Big flash
rBoot Option: RTC data

Booting rom 0.
WAlЊ|тbмтnтp‚Њnnћlnnмђnаpn~мrb’n|bnЊlb`rlpp_task_hdl : 3ffef9a8, prio:14, stack:512
pm_task_hdl : 3ffef410, prio:1, stack:176
frc2_timer_task_hdl:0x3fff4b48, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ May 17 2020 18:10:18
phy ver: 273, pp ver: 8.3

Sysparam OK

WiFi Init
Fatal exception (28): 
epc1=0x40218827
epc2=0x00000000
epc3=0x00000000
excvaddr=0x00000005
depc=0x00000000
excsave1=0x40218824
Registers:
a0 40218824 a1 3fff38d0 a2  00000001 a3  00000000
a4  00000001 a5  00000000 a6  3fff3884 a7  3feffe00
a8  00000000 a9  0000001a a10 00000018 a11 04000102
a12 00000006 a13 3fff2510 SAR 00000008

Stack: SP=0x3fff38d0
0x3fff38d0: 3fff3910 00003002 0000ffff 00000001
0x3fff38e0: 69466957 00000000 00000000 00000000
0x3fff38f0: 00000000 00000000 00000000 00000000
0x3fff3900: 7274616d 32697661 00003233 00000000
0x3fff3910: 00000000 00000000 00000000 00000000
0x3fff3920: 00000000 00000000 00000000 00000000
0x3fff3930: 00000000 00000000 00000000 00000000
0x3fff3940: 00000000 00000000 3fff27dc 00000000

Free Heap: 45936
_heap_start 0x3fff29c8 brk 0x3fff4c90 supervisor sp 0x40000000 sp-brk 45936 bytes
arena (total_size) 8904 fordblks (free_size) 0 uordblocks (used_size) 8904

```

RavenSystem, 2020-05-19:
> Is this the same issue fixed in HAA 2.3.4/OTA 3.0.6?

Yes. You must reflash it.

closed 2020-05-20 by i3laze

i3laze, 2020-05-20:
Yeap,
Taken haaboot.bin release from here https://github.com/RavenSystem/haa/releases
This https://github.com/RavenSystem/esp-homekit-devices/issues/768#issuecomment-581898393 command did the trick. Restored with configuration and HomeKit ID intact.