RavenSystem/esp-homekit-devices #964: homekit icon - garage door > roller shutter

dmagraf, 2020-05-20:
HI!
Thank you very much for your tremendous work in the development of the project!
Now hands have reached the automation of a street sun-protection awning which is controlled only by the RF433 signal. Nebud will go deep into the details, but all my efforts to integrate the RF433 ended in failure. I have been using Sonoff SV for garage doors for a long time. The idea of ​​using Sonoff SV in combination with the remote control got excited. In essence, it turns out that you only need to briefly close the contacts on the remote control to activate the movement of opening and closing the sun blind, and on the rest of the GPIO I can remove the state open / closed. All the original functionality should remain as it is, just replace the hotkey icon from the garage door at least on a roller shutter or something more suitable? Any ideas?
thank
Dmitriy

i3laze, 2020-05-20:
Привет, Dmitriy.

There’s a accessory type Window Covering right for this case. Wiki page:
https://github.com/RavenSystem/esp-homekit-devices/wiki/Window-Covering

I suggest you to try HAA Online Configurator:
https://glumb.github.io/haa-configurator/

Although it doesn’t support window covering yet, try this thing:
1. Paste your Garage JSON configurator into configurator.
2. Replace accessory type “t”:40 to 45.
3. Rename [accessory keys](https://github.com/RavenSystem/esp-homekit-devices/wiki/Garage-Door) like “d”,”c” to relevant window covering keys.

Paste the source and result JSON here for troubleshhpting

dmagraf, 2020-05-20:
Unfortunately, the option you proposed doesn’t work for me, the accessory didn’t hit the homkit. maybe somewhere an error?
`{
  "c":{"l":13,"b":[{"g":0,"t":5}]},
  "a":[{
    "t":45,
    "o":18,
    "0":{"r":[{"g":12,"v":1,"i":0.5}]},
    "1":{"a":0},
    "2":{"r":[{"g":12,"i":1.5},{"g":12,"v":1,"i":0.5},{"g":12,"v":1,"i":2}]},
    "3":{"a":2},
    "f3":[{"g":14,"t":0}],
    "f4":[{"g":14}]
  }]
}`

closed 2020-06-20 by stale[bot]