RavenSystem/esp-homekit-devices #967: Problem with reboot on config

sorriso93, 2020-05-22:
Hello probably I'm missing something to do...
I flashed a couple of wemos d1 mini with relay and today also an electrodragon (ESP826612f), and all have the same behavior, I cannot access the boot config.

flashed with this config, I see it in homekit and the accessory are good. If I try with the physical botton or the logical one, no way to enter the config mode (although I see the wifi HAA but no way to access it).

below the log for the electrodragon, connected to homekit, below the json (but I have the same problem with different esp with relays or temp sensor doing ok on homekit)

{"c":{"q":850,"n":"STRCUCTER","b":[{"g":0,"t":5}]},"a":[{"t":30,"r":15,"g":13,"v":12,"w":14,"b":[{"g":0}]},{"i":0.5,"1": {"s":[{"a":1}]}}]} 

ESP-Open-SDK ver: 0.0.1 compiled @ May 21 2020 19:19:10
phy ver: 273, pp ver: 8.3

Sysparam ready
mode : sta(d8:bf:c0:f9:ca:61)
add if0
WiFi config init
Saved BSSID: 30b5c262a76d
WiFi Mode: Normal
Hostname: STRCUCTER
scandone
add 0
aid 37
cnt 

connected with MAXNETHOME, channel 4
dhcp client start...
ip:192.168.0.86,mask:255.255.255.0,gw:192.168.0.1
Free Heap: 45840
Free Heap: 37256
mDNS announcement: Name=HAA-F9CA61 md=RavenSystem HAApv=1.1id=D1:80:E5:6F:15:2Ec#=8390s#=1ff=0sf=0ci=1sh=jz1OOA== Port=5556 TTL=4500

as soon as I click on homekit on the logical button, it reboot in setup mode but no way to load the config web page or join the wifi from an iphone, an ipad, an android phone or a mac!

Rebooting...

rm match
del if0
usl
sul 0 0

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 2292, room 16 
tail 4
chksum 0x57
load 0x3ffe8000, len 772, room 4 
tail 0
chksum 0x0b
csum 0x0b

rBoot v1.4.0 - richardaburton@gmail.com
Flash Size:   8 Mbit
Flash Mode:   DOUT
Flash Speed:  40 MHz
rBoot Option: Big flash
rBoot Option: RTC data

Booting rom 0.
pp_task_hdl : 3ffeffa0, prio:14, stack:512
pm_task_hdl : 3ffef890, prio:1, stack:176
frc2_timer_task_hdl:0x3fff4b40, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ May 21 2020 19:19:10
phy ver: 273, pp ver: 8.3

Sysparam ready

Home Accessory Architect v2.3.6
Developed by José Antonio Jiménez Campos (@RavenSystem)

SETUP MODE
WiFi config init
Saved BSSID: 30b5c262a76d
WiFi Mode: Normal
HAA Setup
Enabling auto reboot
Start AP SSID=HAA-F9CA61
Start DHCP server
mode : sta(d8:bf:c0:f9:ca:61) + softAP(da:bf:c0:f9:ca:61)
add if0
add if1
bcn 100
Start DNS server
Start HTTP server
Start WiFi scan
scandone
Hostname: HAA-F9CA61
scandone
reconnect
scandone
add 0
aid 37
cnt 

connected with MAXNETHOME, channel 4
dhcp client start...
ip:192.168.0.86,mask:255.255.255.0,gw:192.168.0.1
bcn 0
del if1
mode : sta(d8:bf:c0:f9:ca:61)
Stop DNS server
scandone
scandone
scandone
scandone
scandone
scandone

RavenSystem, 2020-05-22:
When device, in setup mode, connects to your WiFi network, it disabled AP for security. You must use IP address (http://192.168.0.86) or hostname (http://HAA-F9CA61) to load webIU.

Source: https://github.com/RavenSystem/esp-homekit-devices/wiki/Setup-Mode#device-connection

sorriso93, 2020-05-22:
OK now I understood... you have to connect directly via wifi connection to the ip of the device, hence not using the wifi HAA. Let me say that in the wiki to me was not clear at all...
thanks

WizBangCrash, 2020-05-23:
Hi @sorriso93, I've tried to clarify things a little better in the wiki. Hope this helps.

sorriso93, 2020-05-25:
> Hi @sorriso93, I've tried to clarify things a little better in the wiki. Hope this helps.

Yes, I suspect also other similar issues opened are for the same reason ;-)

closed 2020-05-25 by sorriso93