RavenSystem/esp-homekit-devices #969: Sonoff th16 no temp

tech1357-404, 2020-05-23:
Sonoff th16 temp not comeing through to home kit and is it possible for it to work for cooling rather than heating purposes have tried editing Jason string gpio 12 value to 0 from 1

i3laze, 2020-05-24:
Please post your JSON.

tech1357-404, 2020-05-24:
> Please post your JSON config here and also state the model or photo of Temperature sensor you’re using.

{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"t":25,"s":5,"g":14,"0":{"r":[{"g":12}]},"1":{"a":0},"3":{"r":[{"g":12,"v":1}]},"5":{"a":0}}]}

tech1357-404, 2020-05-26:
> Please post your JSON config here and also state the model or photo of Temperature sensor you’re using.

Sorry model is DHT22 sensor

i3laze, 2020-05-26:
Please re-do your confit using HAA Online Configurator:
https://glumb.github.io/haa-configurator/
Use empty template and just add a Thermostat accessory below.

Consult with Wiki about accessory options - you need input and actions “2” and “4” for cooling off and on.
https://github.com/RavenSystem/esp-homekit-devices/wiki/Thermostat

closed 2020-06-25 by stale[bot]

asbesto01, 2022-02-26:
> You need GPIO14 as sensor input and actions “2”, “4” for cooling off and on, instead of “1”, “3” in your config above.
> 
> I suggest you re-do config using HAA Online Configurator: https://glumb.github.io/haa-configurator/ Use empty template and just add a Thermostat accessory below.
> 
> Consult with Wiki about other accessory options: https://github.com/RavenSystem/esp-homekit-devices/wiki/Thermostat

I have the same error, help please!

TH16 with DS18B20 sensor - the sensor shows 0 degrees in homekit.

This is my config:

{
  "c": {
    "l": 13,
    "b": [
      {
        "g": 0,
        "t": 5
      }
    ]
  },
  "a": [
    {
      "0": {
        "r": [
          {
            "g": 12
          }
        ]
      },
      "1": {
        "r": [
          {
            "g": 12,
            "v": 1
          }
        ]
      },
      "b": [
        {
          "g": 0
        }
      ],
      "s": 0
    },
    {
      "t": 22,
      "g": 14,
      "n": 3,
      "b": []
    }
  ]
}