RavenSystem/esp-homekit-devices #970: Missing Reset Homekit ID

jakudo, 2020-05-23:
Hi,

I had one Sonoff mini with an old HAA firmware (0.7.x). I've just updated to the latest firmware but can't add it to the Homekit. I wanted to "Reset Homekit ID" but there is no option for that... Any idea what am I doing wrong?

Thanks,
Jakub

[image]
[image]

tonysprenk, 2020-05-23:
This screen is initial setup. Select auto Ota updates and the WiFi network and click save. Wait ten minutes for it to download HAA and then add it to HomeKit.

jakudo, 2020-05-23:
Thanks, that worked. Strange that before I didn't need to wait that long. Next time, I have to be more patient :)

closed 2020-05-23 by jakudo

skupu, 2021-04-08:
now it doesn't download...