RavenSystem/esp-homekit-devices #972: SONOFF Basic with BH1750 ambient light Sensor (Not exactly an issue at all)

grivenDE, 2020-05-24:
I wonder if it is possible to integrate the BH1750 ambient light Sensor to use it as a trigger in HomeKit. Since HAA Jason Configurator does not offer ambient light Sensors I don't know how to generate a suitable JASON String for this accessory. Is there a way to add the ambient light Sensor or is anybody using ambient light Sensors with HAA?

CryptonWind, 2020-05-25:
I don't have any experience with this sensor and I don't have it to check. 
In theory if you can get at least a 3.3V from somewhere from the board when you get a specific amount of light, then it can be used as a TRIGGER. Of course with some electronic changes.
As I can see the sensor measure the amount of light and not only if it is light or not. 
I doubt you can use it as a trigger too easy.

grivenDE, 2020-05-25:
The wiring of the Sensor ist well documented. Basically you have four wires (GNG, VCC, SDA, SCL) which connect to GND, VCC, RX and TX of the SONOFF Basic so from the physical part it's not an issue at all. I guess my main hassle is the complete absence of Light Sensors in HAA (I read that Light Sensors are still on the to do list so maybe I am out of luck this time and have to wait until they are supported). The BH1750/SONOFF Basic Combo is widely used in Tasmota setups but Tasmota is not what I am after at all.

CryptonWind, 2020-05-25:
> I guess my main hassle is the complete absence of Light Sensors in HAA (I read that Light Sensors are still on the to do list so maybe I am out of luck this time and have to wait until they are supported). 

I understood from initial question that you want to use it as a trigger and not as a light sensor. 
I believe these are two different things. 

These seems to be all the implemented types:
https://github.com/RavenSystem/esp-homekit-devices/wiki/Accessory-Types

closed 2020-06-25 by stale[bot]