RavenSystem/esp-homekit-devices #974: Unable to connect HHA xxxxx after entering setup mode

tarontop, 2020-05-26:
I have successfully connected to Homekit and realized the control. However, when I entered the setting mode, I could not connect to hha-xxxx, indicating that I could not join the network
[image]

grivenDE, 2020-05-26:
Once installed and registered with your network setup mode will work over the Devices IP instead of connecting to it's own WLAN Network. Try pointing your Browser to your Devices IP while it is in setup mode ;)

tarontop, 2020-05-26:
> 一旦安装并向您的网络设置模式注册，它将可以通过设备IP工作，而不是连接到它自己的无线网络。在设置模式下，尝试将浏览器指向设备IP；）

It's settled. Thank you！

closed 2020-06-25 by stale[bot]