RavenSystem/esp-homekit-devices #975: Can no reset homekit id

karnas99, 2020-05-26:
Hi,

I just flashed sonoff sv to HAA. Problem is that web does not offer reset homekit ID so I can’t add device to homekit. Please help.

[image]

karnas99, 2020-05-26:
Ok After some time page reloaded with all options so I reset homekit id. However now I get another error. 
[image]

Kienz, 2020-05-27:
The HomeKit code of HAA devices is 0218-2017 - in your screenshot you are using the HomeKit code 0941-4037.

karnas99, 2020-05-27:
Thanks for help it is working now!

closed 2020-05-27 by karnas99