RavenSystem/esp-homekit-devices #976: Sonoff 4 Channel Switch Model REV 2

rickdhoy, 2020-05-27:
Hi All,

I have a couple of these ( Sonoff 4CH REV 2) and I have been trying to flash them and add them to my system.

I get through the erase and then the flash ok.

I then get it to boot and gain access to the configuration window adding the Json Script and the access point SSID and the password.

Wait the 7 to 10 minutes but can't seem to then be able to add it to Apple HomeKit. It simply just not appear.

i have tried 3 x with both of the devices I have and the same thing occur.

I have also tried using mains power to run the unit thinking that potentially the 3.3volts is not enough to drive the board.

I just get a blank board no lights and again it does not appear in apple HomeKit to allow me to add it.

Any suggestions please ?

Kienz, 2020-05-27:
Can you post your JSON?

rickdhoy, 2020-05-27:
Hi, here you go. 

Thanks for having look for me.

{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":15}]},"1":{"r":[{"g":15,"v":1}]},"b":[{"g":14}]},{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}]},{"0":{"r":[{"g":5}]},"1":{"r":[{"g":5,"v":1}]},"b":[{"g":9}]},{"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"b":[{"g":10}]}]}

Kienz, 2020-05-27:
The JSON is correct - I used the same a few days ago. Did you check your router if the device is connected and/or if it's downloading data?
- Wait longer - maybe bad wifi signal
- Reboot device after eg. 15 minutes

I can successfully install HAA on Sonoff 4CH Pro R2 and add it to HomeKit.

rickdhoy, 2020-05-28:
Ok, thanks. All the other devices are working well and have good connectivity.

I will leave them for 15 minutes or longer and see what happens.

Thanks

Rick

appljd, 2020-06-16:
@Kienz I just flashed a Sonoff 4CH Pro R2. Question: Did you use fullhaaboot.bin to flash ?
My flash shows as succesful. But then when I reboot the device I never see it show up under WiFi selector. What am I missing ? Do I have to preset the Wifi ?

Command: esptool.py --port /dev/cu.usbserial-AD0K64IQ --baud 115200 --after no_reset write_flash --flash_mode dio 0x00000 /Downloads/fullhaaboot.bin --erase-all

esptool.py v2.6
Serial port /dev/cu.usbserial-AD0K64IQ
Connecting....
Detecting chip type... ESP8266
Chip is ESP8285
Features: WiFi, Embedded Flash
MAC: c9:2b:91:40:5c:a2
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 1MB
Erasing flash (this may take a while)...
Chip erase completed successfully in 3.5s
Flash params set to 0x0220
Compressed 399388 bytes to 283241...
Wrote 399388 bytes (283241 compressed) at 0x00000000 in 25.1 seconds (effective 127.1 kbit/s)...
Hash of data verified.

Kienz, 2020-06-17:
I'm using the following command:
```
esptool.py -p /dev/tty.usbserial-AO007UUU --baud 115200 write_flash -fs 1MB -fm dout -ff 40m 0x0 "/Downloads/rboot.bin" 0x1000 "/Downloads/blank_config.bin" 0x2000 "/Downloads/haaboot.bin"
```

appljd, 2020-06-20:
@Kienz Thank you. changing to "dout" did the trick. Also I used an external 3.3v power supply as I believe the power provided via USB might not have been sufficient.
[image]

esptool.py --port /dev/cu.usbserial-AD0K64IQ --baud 115200 --after no_reset write_flash --flash_mode dout 0x00000 /Users/jdman/Downloads/fullhaaboot.bin --erase-all
Using the NodeMCU-PyFlasher-4.0 on macOS Catalina 10.15.5 if that helps anyone else.

closed 2020-07-05 by RavenSystem

ken1170503, 2021-08-25:
> Hi, here you go.
>
> Thanks for having look for me. [...]

> them for 15 minut

I have the same question. 
Could this jason work on "Sonoff 4CH " ?  It is not Sonoff 4CH Pro R2