RavenSystem/esp-homekit-devices #977: Tuya W600-based socket

bj-tracer, 2020-05-27:
Hello!
I bought a new Tuya smart socket in new form factro and with energy monitoring.
Like this https://aliexpress.ru/item/4000478798085.html
But it's based on esp-compatible module with WinnerMicro W600 controller (http://www.winnermicro.com/en/html/1/156/158/497.html).

Can it flashing with HAA?

tonysprenk, 2020-05-27:
No, maybe it’s possible to replace the module with an ESP8266XX module. Can you post pictures of the module inside?

bj-tracer, 2020-05-29:
Hi!
I haven’t deep disassembling of it now.
Only this pictures:
[image]
[image]

Module docs: https://docs.w600.fun/?p=product/tw-02.md

tonysprenk, 2020-05-29:
It’s a winnermicro w600 chip. As far as I know there is no esp8266 equivalent to this module. What you can do is get an esp01 or similar and connect it to the 3.3v, gnd and gpio pins with small wires. Probably not worth the hassle though.

bj-tracer, 2020-06-01:
I know it.
But people flash it with Tasmota firmware by OTA as I know.

bj-tracer, 2020-06-05:
Ok
I replaced module to esp-M3 and flashed HAA.
It's work! :)

tonysprenk, 2020-06-05:
Nice! Do you have any pictures of the process? Or before and after pictures?

bj-tracer, 2020-06-07:
It is the first sample. I tried to use grio2 to control the relay. But it’s not working.

 [image]
[image]
[image]
[image]

And I changed gpio2 to gpio12.

closed 2020-07-07 by stale[bot]