RavenSystem/esp-homekit-devices #978: Json - inching time, button in homekit doesn't change back

karnas99, 2020-05-27:
Hi,

sorry I am quite poor with Json config so I would like to ask for help.

I have sonoff SV with HAA. I want to set 7 sec pulse so I added inching time. 

{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1,"i":7}]},"b":[{"g":0}]}]}

It is working after - 7 sec GPO toggle back however button in home application stays activated.
Is there anything else needs to be added to JSON to get button in home application working properly? Thanks

karnas99, 2020-05-27:
I found it in old thread

closed 2020-05-27 by karnas99

shanghaix, 2020-06-03:
Could you please share your config? I'm looking to make a similar config.

karnas99, 2020-06-03:
{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"i":7,"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1,"i":7}]},"b":[{"g":0}]}]}
 Inching time in this case 7 sec has to be in the root of accesory, then it works properly, tested on sonoff sv