RavenSystem/esp-homekit-devices #979: Sonnoff mini electric sensor Json config

karnas99, 2020-05-27:
Hi,

I am not much familiar with JSON. I have sonoff mini with HAA and I want to build sensor. Once S1 S2 are closed I would like to get notification in home application. I believe that it is necessary to add it as some kind of sensor not switch. Could anybody provide me with JSON config?
Thanks!

karnas99, 2020-05-30:
OK I have certain progress, I added contact sensor to the config. I can see contact sensor in home. IF S1 S2 is connected sensor button is activated. However if S1 and S2 is disconnected button is still activated. I need button deactivated. Here is whole JSON.

{"c":{"b":[{"g":0,"t":5}],"l":13},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0},{"g":4},{"g":4,"t":0}],"s":0},{"t":5,"s":0,"f0":[{"g":0,"t":1}],"f1":[{"g":4,"t":1},{"g":4,"t":0}]}]}

Can anybody advise me how to modify part "s" ?

karnas99, 2020-05-31:
OK I figured out, If you want to use sonoff mini as a contact sensor on S1 S2 input it is quite simple.

{"c":{"b":[{"g":0,"t":5}],"l":13},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0},{"g":4},{"g":4,"t":0}],"s":0},{"t":5,"f0":[{"g":4,"t":1}],"f1":[{"g":4,"t":1},{"g":4,"t":0}]}]}

karnas99, 2020-05-31:
I have one more question my only problem now is that after power cycle home application always show contact sensor open, even if s1 s2 are disconnected.

I tried to add "s":5 but it doesn't work

{"t":5,"s":5,"f0":[{"g":4,"t":1}],"f1":[{"g":4,"t":1},{"g":4,"t":0}]}]}

Is there any way how sensor show according to s1 and s2 after reload?

i3laze, 2020-06-02:
Sorry that you were left playing solo here.
Try https://glumb.github.io/haa-configurator/ for better troubleshooting.

Regarding sensor status - you might try replacing “f0”,”f1” with “g0”,”g1”.
Sensor for some reason currently don’t support initial state. There’s an issue for that https://github.com/RavenSystem/esp-homekit-devices/issues/936

closed 2020-07-02 by stale[bot]