RavenSystem/esp-homekit-devices #980: garage door configuration

Kosmalev, 2020-05-27:
good afternoon.  please help, I use this Jonson config, perhaps help redistribute its status indication from normally open contacts to normally closed and the icon in HomeKit from a sliding gate to a sliding one.  I can not figure out the configurator, thanks in advance

 {"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12,"v":1,"i":0.5}]},"1":{"a":0},"2":{"r":[{"g":12,"i":1.5},
{"g":12,"v":1,"i":0.5},{"g":12,"v":1,"i":2}]},"3":{"a":2},"t":40,"d":31,"f3":[{"g":14,"t":0}],"f4":
[{"g":14}],"f2":[{"g":4,"t":0}],"f5":[{"g":4}]}]}

i3laze, 2020-05-30:
Hi! Try using Online HAA Configurator.
Just copy&paste your JSON and tune it further.
https://glumb.github.io/haa-configurator/

Looks like you missed "1" action. Consult with wiki page:
https://github.com/RavenSystem/esp-homekit-devices/wiki/Garage-Door

I actually suggest you to use an existing template in Configurator instead fo your self-made one above.

Kosmalev, 2020-05-30:
good afternoon.  Thank you very much, I will definitely try.  I took this template from the wiki, it completely works for me, only the indication is made on normally open contacts, and I have normally closed ones, I need to redo this

i3laze, 2020-05-30:
You just change “Single activation” to “Single Activation. Falling Edge“ or back to invert the status.
Do have two sensors GPIO4 and GPIO14 or just one?

Kosmalev, 2020-05-30:
Thank you so much for your help, I'll try and unsubscribe tomorrow

Kosmalev, 2020-05-31:
Thank you very much for your help, everything worked out.  I thought you need to fully compose the configuration, but it turned out that you can edit.  it remains to figure out how to change the gate icon

closed 2020-06-30 by stale[bot]