RavenSystem/esp-homekit-devices #981: Apache License 2.0 incongruities

NedScott, 2020-05-28:
The license for this project says Apache License 2.0, and you have a copy of that license, but you've modified it and added two terms not found in the actual Apache License text:

```
10. Although already forbidden by the sources and subsequent licensing,
      it is not allowed to use or distribute this software for a commercial purpose.

  11. The pairing code is 021-82-017, and can not be changed or modified.
```

You can't call it Apache License 2.0 and also forbid commercial usage. It's also really weird that you would try to add a term for the pairing code in the copyright license, but saying that something like that cannot be changed is also not compatible with the Apache License.

RavenSystem, 2020-05-28:
If you don't like license terms, don't use this software.

NedScott, 2020-05-28:
It's not a complaint. Right now your project isn't Apache License 2.0, it's a default "all rights reserved", basically closed source. If you want it to be non-commercial, but open, that's fine, but you will need to select a different license.

rubinlinux, 2020-05-28:
You can of course license your own work however you want, just be clear and straightforward. If its a modified apache2, say that instead of apache2. It is misleading for people to look at your project and think "Oh, its Apache 2.0 licensed, I like that license" and only after digging deeper find out that it isn't. If you want a non-commercial license, there are some available. The apache2.0 license is approved by the OSI which carries weight for some users. Your modified license has not been.

The top of all of your source files references the official apache2.0 license by URL, while the repository LICENSE file contains a license that claims to be apache2.0 but is not. This is inconsistent and confusing (Which applies? File headers should override LICENSE file, no?)

Also, separately, Wolfssl uses a GPL2 license. Did you obtain a commercial license to allow you to re-license it? Otherwise it seems like distributing binaries would violate their copyright and puts this whole thing into legal grey area unless all your code is GPL2.

RavenSystem, 2020-05-28:
This is the reason because it must be non-commercial:

"HomeKit Accessory Protocol (HAP) is Apple’s proprietary protocol that enables third-party accessories in the home (e.g., lights, thermostats and door locks) and Apple products to communicate with each other. HAP supports two transports, IP and Bluetooth LE. The information provided in the HomeKit Accessory Protocol Specification (Non-Commercial Version) describes how to implement HAP in an accessory that you create for non-commercial use and that will not be distributed or sold."

Term 10 is only a reminder of that.

I will change license name to avoid confusion.

xrust83, 2020-05-28:
why you attacked a person.  The apple does not need needleworkers who will gather in a garage.  collect and sell.  I am sure that this is a direct requirement for the author of the code.  You can do this to pay for the work of the author and the apple.  This license simply prohibits you from earning on someone else’s mind.

rubinlinux, 2020-05-28:
Where is anyone being attacked? The issue tracker is for tracking issues. There was found to be an issue, and it was reported. Sounds like RavenSystem is going to correct it, so the situation improves and we all win as a community.

NedScott, 2020-05-28:
Feel free to do whatever works, but Apple's terms of use for an API and the copyright license for code are two different things. IANAL, but I believe there is no burden on you to include Apple's non-commercial terms in the copyright license for this code. That burden is on the end user and how they use the code.

closed 2020-06-27 by stale[bot]