RavenSystem/esp-homekit-devices #984: OTA failing

i3laze, 2020-05-30:
I've got dozen of HAA devices I've flashed from scratch during last month.
However, I've got two of them not updating further:
Sonoff Basic OTA 3.0.2, HAA 2.1.1
Sonoff S20 outlet OTA 3.0.7, HAA 2.3.5
Here's the output of the second one:

By the way, tested Server: github.com/RavenSystem/haa/releases/latest/download   in the browser - got 404. 
How so? Should I try via the local server?

```
ESP-Open-SDK ver: 0.0.1 compiled @ May 19 2020 19:22:54
phy ver: 273, pp ver: 8.3

Sysparam OK

WiFi Init
Saved BSSID: 18d6c770adca
WiFi Mode: Normal

HAA OTA - NORMAL MODE

mode : sta(68:c6:3a:d5:b3:86)
add if0
scandone
no WiFi found, reconnect after 1s
reconnect
scandone
add 0
aid 7
cnt 

connected with WiFi, channel 1
dhcp client start...
ip:10.0.0.8,mask:255.255.255.0,gw:10.0.0.254

HAA Installer Version: 3.0.7

- Server: github.com/RavenSystem/haa/releases/latest/download
- Port:   443
- SSL:    yes

Restarting...
rm match
del if0
usl
sul 0 0

 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x40100000, len 2292, room 16 
tail 4
chksum 0x57
load 0x3ffe8000, len 772, room 4 
tail 0
chksum 0x0b
csum 0x0b

rBoot v1.4.0 - richardaburton@gmail.com
Flash Size:   8 Mbit
Flash Mode:   DOUT
Flash Speed:  40 MHz
rBoot Option: Big flash
rBoot Option: RTC data

Booting rom 0.
pp_task_hdl : 3ffeffa0, prio:14, stack:512
pm_task_hdl : 3ffef890, prio:1, stack:176
frc2_timer_task_hdl:0x3fff3fd8, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ May 19 2020 19:28:02
phy ver: 273, pp ver: 8.3

Sysparam ready

Home Accessory Architect v2.3.5
Developed by Jos√© Antonio Jim√©nez Campos (@RavenSystem)

SETUP MODE
WiFi config init
Saved BSSID: 18d6c770adca
WiFi Mode: Normal
HAA Setup
Enabling auto reboot
Start AP SSID=HAA-D5B386
Start DHCP server
mode : sta(68:c6:3a:d5:b3:86) + softAP(6a:c6:3a:d5:b3:86)
add if0
add if1
bcn 100
Start DNS server
Start HTTP server
Start WiFi scan
scandone
Hostname: HAA-D5B386
```

archiolochos, 2020-06-01:
Same here but with OTA 3.0.8. There is no Update possible for me.

i3laze, 2020-06-02:
I’ve flashed it manually preserving the HomeKit ID and JSON config using [this method](https://github.com/RavenSystem/esp-homekit-devices/issues/768#issuecomment-581898393).

However, here’s a malfunctioning backup .bin, for troubleshooting purposes:
_to be added_

closed 2020-07-05 by stale[bot]