RavenSystem/esp-homekit-devices #986: Disabling an accessory in HomeKit

rblenkinsopp, 2020-05-30:
I'm currently using this with a NetVIP XS-A14 smart socket. This is a combined outlet and RGB device.

I would like to be able to use this as a single Outlet style accessory, with the RGB used as a status for if the device is enabled or not.

Currently, I have this setup as two accessories with the outlet set to notify the lightbulb to turn on and off in response to the switch. Configuration below:

`{"c":{"b":[{"g":5,"t":5}],"o":3},"a":[{"t":2,"b":[{"g":5,"t":1}],"0":{"r":[{"g":13,"v":0}],"m":[{"g":1,"v":0}]},"1":{"r":[{"g":13,"v":1}],"m":[{"g":1,"v":1}]}},{"t":30,"r":0,"g":2,"v":12,"b":[{"g":5}]}]}`

This works great, but I'd love for the lightbulb not to show up in HomeKit as it's really just an outlet with a visual indicator, and I don't want it to be separably controllable.

Is there any way to achieve this? Either to "Hide" the lightbulb accessory, or have a form of status LED that works with the PWM RGB outputs?

i3laze, 2020-06-01:
Hi,
Your config looks pretty nice when pasted into HAA Online Configurator:
https://glumb.github.io/haa-configurator/

I personally stow unnessesary accessories in special “HomeKit rooms” like “System, Setup, Update”.
 However you might just use a configurator above and totally remove the RGB accessory.
If you still want to have LED, you can hook it up to Outlet accessory as one of GPIO outputs (R:0,G:2 or B:12) for On and Off actions.

rblenkinsopp, 2020-06-01:
Yes the configurator is a useful tool (though notably bugged if you try and generate this config with it, and it doesn't support notifications).

Hiding in an unused room is a workable option but not ideal. I can look at the other option you mention, but by understanding was that it doesn't work with the PWM hardware on the LEDS? I've only had luck with the Lightbulb accessory with these LEDS, the status LED option with them does not work correctly (it momentarily flickers, then goes back to its previous colour).

i3laze, 2020-06-01:
Try and error. 
I know nothing about RGB controlllers, but try these GPIOs above as output and optionally try invert it.

rblenkinsopp, 2020-06-02:
Thanks, it works, it's not quite a nice as the Lightbulb solution as you loose all the dimming effects but it does avoid the need for the accessory. Thanks for the input.

closed 2020-06-02 by rblenkinsopp