RavenSystem/esp-homekit-devices #987: nc -kulnw0 45678 (UDP Logger)

i3laze, 2020-05-31:
Does anybody has _netcat_ UDP logger working on MacOS Catalina?

Although my Wireshark sees all UDP Broadcast packets sent by HAAs around, embeded _netcat_ on Mac somehow doesn't listen to broadcast, only direct single cast is displayed. I even tried _socat_ utility and running as root, with no luck:

```
echo "test" | nc -u 127.0.0.1 45678  -> works
echo "test" | nc -u 255.255.255.255 45678 -> doesn't

sudo nc -kulnw0 45678
test
socat -u udp-recv:45678,reuseaddr -
test
```

RavenSystem, 2020-05-31:
I use `nc -kulnw0 45678` from macOS Catalina (now 10.15.5) without any issue (running as normal user).

RavenSystem, 2020-05-31:
Have you any firewall or something similar that blocks unknown multicast network traffic? Maybe a recent update of your router?

i3laze, 2020-06-01:
Thank you for involvement. Your assurance of this netcat working fine is enough.

I believe I know a VPN tool, messing Mitch my MAC connectivity.
I‘ll post the result when I’ll fix it.

closed 2020-06-01 by i3laze