RavenSystem/esp-homekit-devices #989: Sonoff Basic - Ping, but no HTTP

bartgabriels, 2020-05-31:
I recently did some changes on the home-WiFi (Fortinet, replaced the AP's with it's newer version). 

All Sonofff devices recovered connection, but the 1 I have in test with HAA, has "no response" status in Homekit. I removed the device, but cannot readd.

Solution tried:
- Ping works
- HTTP doesn't work
- Re-Adding in Homekit doesn't work
-Move closed to the AP (as it was pretty remote) didn't work
- Reset the device (10sec press on the button): Led blinks fastly, WiFi is visible on the iphone for a shortt period, but cannot connect to it ...

Any otther things I can try, before trying to reflash?

Bart

i3laze, 2020-06-01:
Did you see HAA-xxxx SSID? Maybe it is in Emergency Setup mode.
Anyway try throwing device into this mode by powering it up and resetting the power in 2 seconds.

And yes, reflashing is always an option.

bartgabriels, 2020-06-01:
> Did you see HAA-xxxx SSID? Maybe it is in [Emergency Setup mode](https://github.com/RavenSystem/esp-homekit-devices/wiki/Setup-Mode).
> Anyway try throwing device into this mode by powering it up and resetting the power in 2 seconds.
> 
> And yes, reflashing is always an option.

Yes, I see the SSID, I can connect to it. It opens the "portal" where I can give the WiFi settings again. It connects (DHCP leases out IP address). I can ping the IP, but cannot connect over HTTP.

i3laze, 2020-06-01:
It happened sometimes in prior HAA versions.

Try power cycling and connecting again. Repeat until you get setup page.

However, I Still suggest reflashing.

closed 2020-07-10 by stale[bot]