RavenSystem/esp-homekit-devices #990: Question about WiFi BSSID choices

macjl, 2020-05-31:
Hello,

Does HAA devices choose the BSSID with the better signal strength, when there are
Multiple with same SSID?
If yes, does they rescan WiFi signal, and how often to choose an other BSSID if there is a better one?

I ask these questions, because I have trouble with HAA devices since I add a second AP in my network. I found that some devices might choose an AP that too far, and don’t seems to reconnect to the other one without rebooting them. So I have unconnected Devices, even when they have a good AP near them. The problem seems only happening to HAA, and not other devices.

i3laze, 2020-06-01:
Yeap, according to my experience they do rescan from time to time and maybe even roam once in an hour.
Indeed at the beginning the AP is chosen based on the signal strength.
I suggest you to reduce transmit signal strength on the farthest AP below the nearest APs signal _(-55dBm is below -50dBm)_. Strong remote signal confuses devices.
For this task use a laptop with some WiFi Scanner software, which displays actual signal strength in the room.

Alternatively use Force BSSID and forget.

macjl, 2020-06-01:
For scanning WiFi strength, I use the Airport App in my iPhone, that provides this feature (mostly unknown by users).

I’ve also noticed that HAA sometimes reconnect to the closest AP for time to time. But I suspect it is done at a fixed interval, and not when it loose signal.

The problem I have is when I reboot one of the AP. Then, all devices reconnect to the remaining AP. When the first AP comes back, it takes hours to reconnect to it, even if the signal of the current AP is pretty bad. 

How does force BSSID works? When choosing a BSSID that is not available, does the device connect to other AP with same SSID?

pilot1981, 2020-06-01:
I don't suggest to use force BSSID option, because if AP X fails, all HAA devices connected to AP X don0t connect to other AP (also if they have same SSID)!

macjl, 2020-06-01:
It would be a good alternative if we had a «prefered SSID» option.

closed 2020-07-01 by stale[bot]