RavenSystem/esp-homekit-devices #994: Robotdyn Dimmer

christophermodom, 2020-06-04:
First of all, a huge thanks to Jose for the hard work!  I am somewhat new to all of this, however I have successfully created a light switch that can be controlled by the physical switch and with homekit...it is flawless.  The configuration for a dimmer is way beyond my capability.  Any chance of creating a config for the Robotdyn AC dimmer (https://robotdyn.com/ac-light-dimmer-module-1-channel-3-3v-5v-logic-ac-50-60hz-220v-110v.html)?  For reference, here is the official Robotdyn repository: (https://github.com/RobotDyn/AC-Light-Dimmer-Module-1-Channel-3.3V-5V-logic-AC-50-60hz-220V-110V)

Thanks!

Santiago512, 2020-06-13:
Following.......

dgauche, 2020-06-16:
Also following, I have 2 of the same Robotdyn devices waiting to be installed.

tokamac, 2020-07-05:
This Robotdyn AC dimmer module does not seem to have an ESP8266 chip (?) which HAA runs on. So it is not compatible. 

Besides this main issue, considering other solutions having a separate board with ESP8266 (e.g. Arduino-based), see [this issue](https://github.com/arendst/Tasmota/issues/5573#issuecomment-479976781) as to why it has not been done (and won't) with Tasmota. Although it is true that Tasmota can be considered as a bloatware (due to its many concurrent protocols running very low on memory on a quite underpowered chip) compared to lightweight (HomeKit-only) HAA, I wonder if HAA is considered "real time".

closed 2020-07-05 by RavenSystem

christophermodom, 2020-07-07:
The Robotdyn AC Dimmer can be controlled by an ESP8266 or ESP32.  I was hoping to control it with a Wemos D1 Mini or an ESP8266 Node MCU.  Controlling this dimmer was the most requested feature for a while over at ESPHome.  They (I) have it working, but I'm seeing some occasional flickering.

Thanks

jmgutierrezc1, 2023-05-15:
"c":{"io":[[[3],1],[[2],7,0,0,2],[[0],6]],
"q":110,"zc":[3,2]},
"a":[{"t":30,"g":[2],
"b":[{"g":0}],"e":5,"d":0.2,"st":384}]}

the only porblem that I have not been able to fix is that @ 43% the bulb turns off,, and there are ocasionla flickers especially @ low brightness