RavenSystem/esp-homekit-devices #998: Problem on water valve on 2.4.5?

sorriso93, 2020-06-07:
Hello after several flash on my wemos d1 mini I suspect that;

{"c":{"o":0,"n":"TERCAMIRRIG","l":13,"i":1},"a":[{"0":{"r":[{"g":5,"v":1}]},"1":{"r":[{"g":5,"v":0}]},"t":20,"b":[],"s":0,"w":0,"d":3600}]}

this configuration runs with no problem on 2.3.10, but on 2.4.5 it doesn't even start the accessory (on the log it stops on en1 interface)

i3laze, 2020-06-08:
Please, post current log output.
Then enable UART (USB Serial) or Network (netcat UDP Broadcast) logging and post extended output.

sorriso93, 2020-06-11:
> Logs, please.

Ok I'm troubleshooting with 231 the wemosd1mini+ dht + PIR. The goos news is the now the PIR immediatly reacts the bad news is that after it activate I see a reboot in setup mode. I discovered it looking at the serial. How can I log via netcat UDP Broadcast? I didn't find any info in the wiki.
Tomorrow I will also check the water valve configuration with wemos d1 mini + relay

sorriso93, 2020-06-11:
No more logs for 245 than this as it is blocked and does not start even the HAA wifi... So I cannot configure it and use UART or network log collection...

pm_task_hdl : 3ffef890, prio:1, stack:176
frc2_timer_task_hdl:0x3fff4b48, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ Jun  6 2020 23:59:45
phy ver: 273, pp ver: 8.3

Sysparam ready
mode : sta(84:f3:eb:25:7a:7d)
add if0

i3laze, 2020-06-11:
> the config is ok on 2310. on 2.4.5 it is stuck on this. Also another wemos d1 mini + dht and PIR have the same behaviour (ok on 2310, same problem with 245)
> 
> Home Accessory Architect v2.4.5
> Developed by José Antonio Jiménez Campos (@RavenSystem)
> 
> SETUP MODE
> WiFi config init
> No WiFi config found
> Start AP SSID=HAA-257A7D
> Start DHCP server
> mode : sta(84:f3:eb:25:7a:7d) + softAP(86:f3:eb:25:7a:7d)
> add if0

Can’t you see you’re in setup mode here :) it won’t respond to Home.app in this mode, only HTTP.
You still need to add UART logging, that will display all the actions happening:
`{"c":{"o":1},"a":[`

sorriso93, 2020-06-11:
ok. flashed a brand new wemos d1 mini.
The first boot is ok, I get HAA network and I put this configuration json that is ok on 231:

`{"c":{"n":"SBMAX","o":3},"a":[{"t":12,"f1":[{"g":0,"p":0}],"s":0,"i":5},{"t":24,"b":[],"g":2,"n":2,"z":-2,"h":5}]}`

then the restart and this is the complete log. I cannot reach the web page config and it get stuck as the previous one.

```
load 0x40100000, len 2292, room 16 
tail 4
chksum 0x57
load 0x3ffe8000, len 772, room 4 
tail 0
chksum 0x0b
csum 0x0b

rBoot v1.4.0 - richardaburton@gmail.com
Flash Size:   8 Mbit
Flash Mode:   DOUT
Flash Speed:  40 MHz
rBoot Option: Big flash
rBoot Option: RTC data

Booting rom 0.
pp_task_hdl : 3ffeffa0, prio:14, stack:512
pm_task_hdl : 3ffef890, prio:1, stack:176
frc2_timer_task_hdl:0x3fff4b48, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ Jun  6 2020 23:59:45
phy ver: 273, pp ver: 8.3

Sysparam ready

Home Accessory Architect v2.4.5
Developed by José Antonio Jiménez Campos (@RavenSystem)

SETUP MODE
WiFi config init
Saved BSSID: 30b5c262a76d
WiFi Mode: Normal
HAA Setup
Enabling auto reboot
Start AP SSID=HAA-E3E972
Start DHCP server
mode : sta(84:f3:eb:e3:e9:72) + softAP(86:f3:eb:e3:e9:72)
add if0
add if1
bcn 100
Start DNS server
Start HTTP server
Start WiFi scan
scandone
Hostname: HAA-E3E972
scandone
reconnect
scandone
add 0
aid 46
cnt 

connected with MAXNETHOME, channel 4
dhcp client start...
ip:192.168.0.52,mask:255.255.255.0,gw:192.168.0.1
bcn 0
del if1
mode : sta(84:f3:eb:e3:e9:72)
Stop DNS server
scandone
scandone
{l d⸮⸮< ⸮$⸮<⸮d⸮c|⸮⸮⸮⸮s⸮c⸮#⸮⸮gn⸮l'g⸮⸮⸮#x⸮⸮ds${lp⸮o⸮⸮lc'⸮<⸮⸮d⸮c⸮⸮o'⸮ $⸮⸮d`⸮ng⸮$or⸮ۛg⸮⸮dp⸮'⸮{⸮⸮⸮⸮⸮cg⸮|⸮⸮x⸮⸮go⸮ ⸮l`⸮'gd`or⸮ۛg⸮⸮`;⸮⸮'ě`s$⸮c⸮ng⸮⸮⸮x|⸮#$c⸮gp`#⸮c⸮#⸮⸮'g⸮⸮o⸮c⸮ncx⸮ol⸮g⸮⸮o⸮o~#dlpc⸮⸮ ⸮g⸮⸮odl'⸮⸮⸮⸮d⸮⸮⸮`⸮o⸮⸮'⸮⸮ۃb⸮$dl~rd⸮b⸮go⸮⸮l⸮'gn ⸮g⸮⸮⸮'⸮⸮gd⸮b⸮go⸮⸮l⸮'o' ⸮⸮` ⸮⸮⸮⸮`s$⸮'g⸮⸮<⸮⸮⸮g⸮ ⸮#$pp_task_hdl : 3ffeffa0, prio:14, stack:512
pm_task_hdl : 3ffef890, prio:1, stack:176
frc2_timer_task_hdl:0x3fff4b48, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ Jun  6 2020 23:59:45
phy ver: 273, pp ver: 8.3

Sysparam ready
mode : sta(84:f3:eb:e3:e9:72)
add if0
;d l⸮⸮| ⸮l⸮|⸮d⸮c|⸮⸮⸮⸮s⸮c⸮c⸮⸮o'⸮$go⸮⸮⸮cp⸮⸮lrds$8⸮g⸮⸮lc'⸮|⸮⸮l⸮c⸮⸮'g⸮ l⸮⸮l ⸮'g⸮d';Ǜ⸮o⸮⸮$x⸮o⸮{⸮⸮⸮⸮⸮cg⸮|쎇p⸮⸮o'⸮ ⸮$`⸮o'$`os⸮⸮⸮gē`;⸮⸮o⸮⸮`{l⸮c⸮o'⸮⸮⸮x|⸮#dc⸮op`c⸮c⸮c⸮⸮gn⸮⸮'⸮c⸮gb8?⸮'$⸮o⸮⸮'⸮⸮gc$$8#⸮⸮`⸮n⸮⸮'l$g⸮⸮⸮⸮l⸮⸮⸮ ⸮'⸮⸮g⸮ۓ⸮c⸮lld{l⸮c⸮o'⸮⸮d⸮o'g ⸮'⸮⸮⸮o⸮⸮n$⸮c⸮'g⸮⸮l⸮'go ⸮⸮` ⸮⸮⸮⸮`{l⸮og⸮⸮|⸮⸮⸮g⸮ ⸮b$pp_task_hdl : 3ffeffa0, prio:14, stack:512
pm_task_hdl : 3ffef890, prio:1, stack:176
frc2_timer_task_hdl:0x3fff4b48, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ Jun  6 2020 23:59:45
phy ver: 273, pp ver: 8.3

Sysparam ready
mode : sta(84:f3:eb:e3:e9:72)
add if0
```

RavenSystem, 2020-06-11:
In setup mode, I don't see any issue. You must be able to connect to device.

In normal mode, logs are necessary to know where is the problem. Activate/capture them, please.

i3laze, 2020-06-11:
Yeah, for some reason you’re still in Setup Mode (HTTP only),  not Normal Mode (HomeKit only).
Open webpage, click Save and then post the log again.

sorriso93, 2020-06-11:
> In setup mode, I don't see any issue. You must be able to connect to device.
> 
> In normal mode, logs are necessary to know where is the problem. Activate/capture them, please.

Probably I didn't explain very well. 
1) flashed the wemos. I tried both fullhaa and the haamain 2.4.5.
2) setup ok and put the json with UART and network logs activated (o = 3), as reported in previous post
3) reboot
4) the only log I get with UART enabled, it is stuck on add if0

ESP-Open-SDK ver: 0.0.1 compiled @ Jun  6 2020 23:59:45
phy ver: 273, pp ver: 8.3

Sysparam ready
mode : sta(84:f3:eb:e3:e9:72)
add if0

5) no way to access the webpage via 192.168.0.x and no HAA wifi. Same behaviour with 3 wemos d1 mini and the two different json (water valve and DHT+PIR), both with uart enabled on serial and network.

RavenSystem, 2020-06-11:
o = 3 is network log only. Then, nothing more than `Sysparam ready mode : sta(84:f3:eb:e3:e9:72) add if0` will be showed by UART.

sorriso93, 2020-06-11:
> o = 3 is network log only. Then, nothing more than `Sysparam ready mode : sta(84:f3:eb:e3:e9:72) add if0` will be showed by UART.
> 
> https://github.com/RavenSystem/esp-homekit-devices/wiki/General-Configuration#Log-output

You are right. Before I used o=1 but the risult was the same.
Anyway, I reflashed both wemos, this round two times with the erase_flash parameter. This has been the only difference in the last reinstall I made... and incredible but true now it is all ok both the water valve and the pir+dht json config.
Starting from yesterday I flashed almost 10 times both wemos with the same failure on 245  vs 231 correctly booting and added on homekit. I erased the flash before loading HAA in both, but probably repeating two times the erase_flash is the correct way to do it!
I think it can be closed the issue
Many thanks

i3laze, 2020-06-11:
Welcome,
However one erase_flash is totally enough.

sorriso93, 2020-06-11:
> Welcome,
> However one erase_flash is totally enough.

I fully agree, but it is the only difference between the last flash on both wemos, and the 20 I made before...

closed 2020-06-12 by sorriso93