RavenSystem/esp-homekit-devices #1002: Garage door example - comprehension question about actions

MartinGaeumann, 2020-06-09:
Hi all

I have a question of understanding about the following JSON-line, which is taken from the garage door example: 
 "2":{"r":[{"g":12,"i":1.5},{"g":12,"v":1,"i":0.5},{"g":12,"v":1,"i":2}]},

In my view: 
- GPIO12 is set to 0 for 1.5sec
- then GPIO12 is set to 1 for 0.5sec
- then GPIO12 is set to 1 for 2sec?

The device (Sonoff SV) reacts correctly: 
- one pulse (relay closing) to stop the garage door
- short break
- one pulse (relay closing) to start the garage door in the other direction

I can't figure out how the device reacts to the code.
Can someone explain this to me?

Best regards
Martin

tokamac, 2020-07-04:
I don't understand neither why there are three consecutive activations of the relay for the opening command (and the last two are even two consecutive ON states aka "Drive output high").

But this is not the sole issue in the wiki about Garage door. The [Actions](https://github.com/RavenSystem/esp-homekit-devices/wiki/Garage-Door#Actions) table in the page states that action "**1**" is to open a closed door, but in the provided code this action is defined in "**2**". 

Above this table, the "Garage Door Example" part lists the four actions, where we can read: 
> door is closed and receives opening order ("2"), door is opening and receives closing order ("2")

so they have the same number! But due to the discrepancy between the Actions table and the code example, it is impossible to guess which one is really "**1**" or "**2**"…

This make me wonder if anyone has ever tried and succeeded with this, and why there is no working solution provided in the various posts yet.

RavenSystem, 2020-07-05:
@tokamac I think your explanation is good.

> door is closed and receives opening order ("2"), door is opening and receives closing order ("2")

Now it is fixed.

closed 2020-07-05 by RavenSystem

tokamac, 2020-07-05:
Thanks but there seems to be another discrepancy with the sentence:

> When the door is **fully closed** "f3" is triggered and then the door is **fully open** "f4" is triggered. GPIO 14 is used to indicate this state.

See the definition of "f4" in the [State and Status Inputs](https://github.com/RavenSystem/esp-homekit-devices/wiki/Garage-Door#State-and-Status-Inputs) table:

> "f4" | Indicates that garage door is **opening**

Because **fully open ≠ opening**

WizBangCrash, 2020-07-09:
> 
> > "f4" | Indicates that garage door is **opening**
> 
> Because **fully open ≠ opening**
> Thus either the code should trigger "f2" instead of "f4", or some definitions in the State and Status Inputs table are incorrect.

@tokamac Well spotted :-)
I have updated the wiki to reflect that GPIO 14 indicates the closed state of the door.