RavenSystem/esp-homekit-devices #1008: Motion Sensor

Przylepzwm, 2020-06-11:
Hello, can anyone help me with JSON configuration for motion sensor.  After configuring two motion sensor accessories, I added them to the Home application.  Unfortunately, the software interprets every second signal as motion on the configured GPIO.  To eliminate the problem on the sensors side, I connected the buttons - the same: every second press activates the LED on the output (GPIO 16) and notification in the Home application.  
Attempts to change JSON have already taken me two days, because entering Setup Mode can be difficult. 
Ultimately, I want to have 3 motion sensors connected to one ESP8266 (nodemcu-v3-lolin ) and use them as separate accessories in Home.  Any high state on GPIO INPUTS should be "motion detected".
I am currently using the following configuration: 
{
  "c": { "l": 13},
  "a": [
{
    "t": 12,
    "i": 10,
    "0": { "r": [{ "g": 16, "v": 0 }] },
    "1": { "r": [{ "g": 16, "v": 1 }] },
    "f1": [{ "g": 4 }]
},
{
    "t": 12,
    "i": 10,
    "0": { "r": [{ "g": 16, "v": 0 }] },
    "1": { "r": [{ "g": 16, "v": 1 }] },
    "f1": [{ "g": 5 }]
},
{
    "t": 12,
    "i": 10,
    "0": { "r": [{ "g": 16, "v": 0 }] },
    "1": { "r": [{ "g": 16, "v": 1 }] },
    "f1": [{ "g": 14 }]
}

]
}

Unfortunately now, it is not possible to add the device to HOME - "the accessory is already added" which may indicate a wrong JSON. Any idea? Thanks

i3laze, 2020-06-13:
Firstly, I suggest you to add these two system accessories, which will allow you to enter setup mode and update the device right from the a Home.app:
`,{"1":{"s":[{"a":1}]}},{"1":{"s":[{"a":2}]}} `

Secondly you’ve done exact config as [in the Wiki](https://github.com/RavenSystem/esp-homekit-devices/wiki/Sensors), so I really have no idea why its triggered only every second time.

Be aware that cool down period for the sensor you set is 10 sec.
Try adding a pullup option or inverting input or changing trigger to falling edge... 
This Online Configurator should help you https://glumb.github.io/haa-configurator/

Lastly, attach a serial cable and enable UART logging (“o”:1), may be you’ll get some hint.

tlagarto, 2020-06-13:
Hi @i3laze, why the two setup buttons?
I use one with this Json:`{"1":{"s":[{"a":1}]}}`
Why the second one? What does the `"a":2` string do?

i3laze, 2020-06-13:
It goes OTA update directly. Useful if you have bunch of devices and @new firmware released.

tlagarto, 2020-06-13:
That's cool. I'll use it.
Just one more question, on the Home app how do I know which switch is the setup and which is the update?

i3laze, 2020-06-13:
That’s the tricky one :)
First one should be Setup... however I just try one of them:
Either it goes to setup mode in 10 sec and webpage opens... or it doesn’t and then OTA cycle finishes in 30-60 sec and device is responsive in Home.app again.

Przylepzwm, 2020-06-14:
Thank you for answer!
 The UART logging advice was very useful, and the online configurator is great.  Thanks to this, I was able to find out the reason for the inability to add to the Home App.
I discovered an interesting thing.  Despite erasing the flash with the bin file (blank_1mb from 0x1 .. - 0x4 ..) there is still old data.  First start after flashing, and the device tries to connect to a network that was configured a few days earlier.  The Reset ID also works strangely.  After entering Setup mode, pasting a new JSON and clicking RESET ID on restart, the device read some old JSON (not used recently). How can I erasing flash memory completely (the easiest way)?

 I was able to add two Virtual buttons.  the one from update works fine, I still have problems with Setup mode (the button looks ok).  I can enter the Setup mode only if it is flashed again or when the WiFi network configured on the device is unavailable. Is it normal ?

As for JSON itself.  I found the same problem in closed issuses:
 https://github.com/RavenSystem/esp-homekit-devices/issues/549
I used their solution and now App shows motion dedection when I have a negative edge. This is better than the second positive, but I would like to have every positive.  What else can I try?

{
  "c": { "o": 1,"i":1,"l": 13},
  "a": [
{"t": 12,"i": 10,"f1": [{ "g":4,"p":0,"i":0,"t": 0 }]},
{"t": 12,"i": 10,"f1": [{ "g":5,"p":0,"i":0,"t": 0 }]},
{"t": 12,"i": 10,"f1": [{ "g":12,"p":0,"i":0,"t": 0 }]}
]}

i3laze, 2020-06-14:
Sorry, can’t help you further.. wait for someone experienced in flashing and motion sensors.

pawelosin, 2020-06-14:
Did you tried to erase memory in your device and flash it again? I am using this command to erase my device (for macOS) esptool.py -p /dev/tty.SLAB_USBtoUART erase_flash

sorriso93, 2020-06-16:
It's not directly related but I also had an issue where only after two sequential flashes on wemos d1 mini I've got 2.45 running correctly. I also used same command on macos

Przylepzwm, 2020-06-22:
> It's not directly related but I also had an issue where only after two sequential flashes on wemos d1 mini I've got 2.45 running correctly. I also used same command on macos

the above code works as a negative edge, so it's a bit better than each second signal.  I will not give up and as soon as I find more time I will try to test various options

Maksut11, 2020-07-02:
Hi, have you solved the sensor problem?

vijaydembla1, 2020-07-19:
Something interesting.

I have been trying a lot to get the motion sensor work but no avail, have tried numerous configs. I started to believe that the 2 motion sensor I have could be faulty, but that wasn't the case.

A few days ago, I raised an issue regarding the contact sensor and received a working solution  as 
{"t":5,"f0":[{"g":5,"t":0}],"f1":[{"g":5}]}, this worked great for the contact sensor, 

similar config for the motion sensor didn't work :
{"t":12,"f0":[{"g":10,"t":0}],"f1":[{"g":10}]}, 

What's interesting is the motion sensor, if connected to the GPIO pins that are configured for Garage door sensor or another door sensor is responding correctly. For eg. if I wave to the motion sensor, the garage door shows its status as open and returns to close, similarly with the door sensor, I wave at the motion sensor, the door shows as open and returns to close in few seconds, this made me believe that the motion sensor is working, something either definitely wrong with my config or with the flash file.
Any help will be appreciated.

sorriso93, 2020-07-23:
what kind of motion sensor do you us

vijaydembla1, 2020-07-24:
PIR sensors
HC sr501

Przylepzwm, 2020-11-02:
Correctly functioning 3 motion sensors and 5 relays on Lolin v3:
{
  "c": { "o": 1},
  "a": [
{"t": 12,"i": 10,"f1": [{ "g":14,"p":0,"i":1,"t": 0 }]},
{"t": 12,"i": 10,"f1": [{ "g":12,"p":0,"i":1,"t": 0 }]},
{"t": 12,"i": 10,"f1": [{ "g":13,"p":0,"i":1,"t": 0 }]},
{"t": 1,"d": 3600,"s": 5,"0": { "r": [{ "g": 16, "v": 0 }] },"1": { "r": [{ "g": 16, "v": 1 }] }}, 
{"t": 1,"d": 3600,"s": 5,"0": { "r": [{ "g": 4, "v": 0 }] },"1": { "r": [{ "g": 4, "v": 1 }] }}, 
{"t": 1,"d": 3600,"s": 5,"0": { "r": [{ "g": 5, "v": 0 }] },"1": { "r": [{ "g": 5, "v": 1 }] }}, 
{"t": 1,"d": 3600,"s": 5,"0": { "r": [{ "g": 3, "v": 0 }] },"1": { "r": [{ "g": 3, "v": 1 }] }},
{"t": 1,"d": 3600,"s": 5,"0": { "r": [{ "g": 1, "v": 0 }] },"1": { "r": [{ "g": 1, "v": 1 }] }}
]}