RavenSystem/esp-homekit-devices #1009: How to flash Sonoff mini to HAA in Mac OS?

tlagarto, 2020-06-11:
I've surfed the wiki and can't find an answer to this one.
I'm sure it's a dumb questions, but I only have a Mac and no Windows PC.
What tools should I use to flash a sonoff mini to HAA?

Thanks!

closed 2020-06-11 by tlagarto

bj-tracer, 2020-06-18:
I soldered wires to the pcb and normally flash with ttl-adapter.
[image]

tlagarto, 2020-06-18:
Yes, I built a jig to flash it without soldering every time.
But I already found a way (command line) to flash the devices.
Thanks anyway.
[image]
[image]