RavenSystem/esp-homekit-devices #1010: Question about http action and influxdb

sorriso93, 2020-06-11:
Hello
I checked in the wiki but I don't understand if it is possible to log temp & umidity in a influx db on a continous base

{
  "c": { "l": 13 },
  "a": [{
    "t": 22,
    "n": 5,
    "j": 10,
    "y0": [{
      "v": -20,
      "r": 1,
      "0": { "h": [{ "h": "influxdb.lan", "p": 8086, "m": 2, "u": "write?db=haadb", "c": "sensor,device=office value=30" }] }
    }]
  }]
}

If I understood correctly it is possible to start an action using, for example, a -20 temperature, the action should be executed every time the temperature is > -20. 
Is it possible to post on the db the temperature measured instead of a fixed value? If yes also for humidity how can I do?

WizBangCrash, 2020-06-14:
Hi @sorriso93,
At the moment it is not possible to post the actual temperature of the sensor, only that a wildcard has been met.

sorriso93, 2020-06-15:
> Hi @sorriso93,
> At the moment it is not possible to post the actual temperature of the sensor, only that a wildcard has been met.

Ah ok... I would it be very useful the possibility to log the hystorical data of all sensors via http post to influx db or similar. Hope it will be included in a next release

WizBangCrash, 2020-06-15:
Yes. So do I :-)

closed 2020-06-15 by WizBangCrash