RavenSystem/esp-homekit-devices #1013: Doorbell configuration

tlagarto, 2020-06-14:
Is there a doorbell configuration?
I'm wondering how to use a sonoff mini as a doorbell but I'm a bit lost on the configuration. 

How should I make it so that homeKit recognise it as a doorbell and provide me with a feed from a Arlo Pro 2 camara (HomeKit compatible)?
I could then add a second Sonoff mini (with a dry contact) to work as a door/gate opener ( with an electric door lock).

Did anyone solved this before?
Suggestions? Ideas?

shanghaix, 2020-06-17:
Homebridge ffmpeg camera does plugin has an option with a doorbell button that you can trigger when someone presses on the sonoff mini, so you get a notification and a photo when someone presses the doorbell.

tlagarto, 2020-06-21:
Currently, I do not have Homebridge installed. I was hoping for a standalone alternative. My guess is that it is not yet possible. But, thanks anyway, I appreciate the suggestion.

tokamac, 2020-07-01:
Is it possible to configure a Sonoff mini as a (fake) motion sensor to HomeKit, instead of a switch? 
If this is possible, the "motion" state would not be caused by a real motion, but when the physical switch is triggered.
So that when the switch is ON, motion sensor is ON. Then it automatically returns to OFF state after a few seconds.

If this is possible, you just have to allow notifications for this accessory exposed as a motion sensor in Home app, and add your Arlo Pro 2 camera in the same room. The room has to be different than the default room. When someone presses the doorbell button, you get a rich notification with a picture from the camera, and the live feed if you touch it.

But I don't know if this is possible (configure a Sonoff mini to mimic a motion sensor).

tlagarto, 2020-07-02:
> Is it possible to configure a Sonoff mini as a (fake) motion sensor to HomeKit, instead of a switch?
> If this is possible, the "motion" state would not be caused by a real motion, but when the physical switch is triggered.
> So that when the switch is ON, motion sensor is ON. Then it automatically returns to OFF state after a few seconds.
> 
> If this is possible, you just have to allow notifications for this accessory exposed as a motion sensor in Home app, and add your Arlo Pro 2 camera in the same room. The room has to be different than the default room. When someone presses the doorbell button, you get a rich notification with a picture from the camera, and the live feed if you touch it.
> 
> @RavenSystem, is it possible to configure a Sonoff mini to mimic a motion sensor and trigger its motion state via a physical momentary switch?

It is possible, I've done it before.
I did not test it with the camera on the same room.
I'll try that. Thanks!

tlagarto, 2020-07-02:
Hi again, I've tried using a sonoff mini to mimic the motion sensor this way:
`{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"t":1,"0":{"r":[{"g":12,"v":0}],"m":[{"g":2,"v":0}]},"1":{"r":[{"g":12,"v":1}],"m":[{"g":2,"v":1}]},"s":5,"b":[{"g":0},{"g":4},{"g":4,"t":0}]},{"t":12},{"1":{"s":[{"a":1}]},"i":0.5,"b":[],"s":0}]}`
But the notification states "motion detected by <room name> <device name>" (not a rich notification with a snapshot of the camera). On clicking the notification, it will show me the device but not the camera feed in the same room.

Am I doing something wrong?
How should I set it up?

tokamac, 2020-07-02:
Just checked the problem on the Internet and it's too bad that its seems Apple removed this feature from iOS 13, whereas it was enabled in iOS 12. Now, apparently the motion sensor has to be a service integrated within the same camera accessory (not another distinct accessory): [https://www.reddit.com/r/HomeKit/comments/dl63do/homekit_notifications_no_longer_include_preview/](https://www.reddit.com/r/HomeKit/comments/dl63do/homekit_notifications_no_longer_include_preview/)

closed 2020-08-02 by stale[bot]